cesanta/mongoose #1468: Mongoose 7.6 client code issue on Ubuntu

kentohub, 2022-02-14:
- My goal is: get client code in Mongoose 7.6 to connect
- My actions were: Upgraded our working application from Mongoose 6.18 to 7.6.
- My expectation was: Server and client code would work at least at socket level
- The result I saw: Server code works ok, client code refuses to connect at all
- My question is: 

As above, on straightforward Linux platform (Ubuntu 18), I've migrated a large application
from Mongoose 6.18 to 7.6. It's mostly in c++, except for Mongoose. For c++ it uses
g++ compiler, for c it's gcc. The build seems completely fine. When I run the application,
acting as a client, Mongoose 7.6 is unable to connect to any server, and there is no
MG_EV_CONNECT event at all. The event handler is being called, I get OPEN and
CLOSE events. I restructured the event handler to work similarly to the http-client
sample code. The symptom is that I get an immediate close event for the connection.

Under the debugger, I stepped into Mongoose code. In mg_connect_resolved(), there is
a connect() call which returns -1 and errno is EINPROGRESS, which is normal.
However the polling loop is supposed to later call connect_conn(), and send the 
CONNECT event, but this never happens! I think connect_conn() is called, but
there is some error return there.

Any idea how this situation can arise? I don't really want to debug Mongoose code.
I did build the http-client sample and it seems to connect ok. I can see no build-related
issues that may cause this. As mentioned 6.18 was working fine for client and server.

# Environment

- mongoose version: 7.6
- Compiler/IDE and SDK: gcc and g++
- Target hardware/board: Ubuntu 18
- Connectivity chip/module: _?_

cpq, 2022-02-15:
@kentohub is your client trying to connect to some remote server ?

cpq, 2022-02-15:
followed quickly by `MG_EV_CLOSE` ?
Please set debug level to 4 and share full log.

kentohub, 2022-02-18:
I've narrowed it down a bit... the select() system call in mg_iotest is always returning 0
(except returns -1 at the end), so c->is_readable and c->is_writable are never set!
So mg_mgr_poll() never calls connect_conn(). Don't know why this is happening.
It also happens on completely different platform, so not just an Ubuntu Linux thing.
But 6.18 also uses select(), and it works!

cpq, 2022-02-19:
@kentohub thanks.
Could you add some extra debugging please - in the mg_iotest, add a trace, which connections are being added:

```c
  for (c = mgr->conns; c != NULL; c = c->next) {
    if (c->is_closing || c->is_resolving || FD(c) == INVALID_SOCKET) continue;
    int w = 0; // <------------------- add this line
    FD_SET(FD(c), &rset);
    if (FD(c) > maxfd) maxfd = FD(c);
    if (c->is_connecting || (c->send.len > 0 && c->is_tls_hs == 0))
      FD_SET(FD(c), &wset),  w = 1;  // <------- add ", w = 1"
      MG_DEBUG(("  %lu %d", c->id, w));  // <---------------- add this
  }
```

To ease debugging, make sure there only one connection preset. You're already disabling TLS and DNS, which is good.

kentohub, 2022-02-22:
Ok here is the code now in mg_iotest.  The MG_DEBUG didn't work, so I used
the LOG macro. I already had some other LOG test code in there, maybe they'll help.
I noted that "socket added to wset" never happens either.

=========================
  for (c = mgr->conns; c != NULL; c = c->next) {
    if (c->is_closing || c->is_resolving || FD(c) == INVALID_SOCKET) continue;
    int w = 0; // <------------------- add this line

    FD_SET(FD(c), &rset);
    LOG(LL_DEBUG, ("socket added to rset: %d", FD(c)));  // TESTING

    if (FD(c) > maxfd) maxfd = FD(c);

    if (c->is_connecting || (c->send.len > 0 && c->is_tls_hs == 0)) {
      FD_SET(FD(c), &wset),  w = 1;  // <------- add ", w = 1"
      LOG(LL_DEBUG, ("socket added to wset: %d", FD(c)));  // TESTING
    }

    LOG(LL_DEBUG, ("DEBUG CONN %lu %d", c->id, w) );  // <---------------- add this
  }

  if ((rc = select((int) maxfd + 1, &rset, &wset, NULL, &tv)) < 0) {
    LOG(LL_DEBUG, ("select: %d %d", rc, MG_SOCK_ERRNO));
    FD_ZERO(&rset);
    FD_ZERO(&wset);
  }

  LOG(LL_DEBUG, ("select returned %d", rc));  // TESTING

  for (c = mgr->conns; c != NULL; c = c->next) {
    // TLS might have stuff buffered, so dig everything
    c->is_readable = FD(c) != INVALID_SOCKET && FD_ISSET(FD(c), &rset);
    c->is_writable = FD(c) != INVALID_SOCKET && FD_ISSET(FD(c), &wset);
  }

[mon2.log](https://github.com/cesanta/mongoose/files/8119864/mon2.log)

cpq, 2022-02-23:
@kentohub thank you.
So the fact that the socket is not added to wset is the problem.
Since I could not reproduce that locally, please continue debugging on your side.
Please show `c->is_connecting`, `c->send.len` and `c->is_tls_hs` in the `socket added to rset` message.
Let's see what's the state of the connection and why it is not addefd to the wset.

kentohub, 2022-02-23:
Ok, I added those, and also added msg in mg_connect_resolved() where
c->is_connecting is set to 1. Results are very strange, the c->id values
don't seem to match up?

[mon3.log](https://github.com/cesanta/mongoose/files/8125951/mon3.log)

kentohub, 2022-02-23:
It looks like in mg_iotest, the loop over mgr->conns only has one
connection in it, since c->id is always 1.
That's why is_connecting is always 0 there, it's always looking
at the initial listening connection (1).
The outgoing client connections are at id 2,3,4... and they never
show up in the loop in mg_iotest.

Is it possible the outgoing connections are not being added to
the linked list? I was even thinking if there was a conflict between
the code macros like LIST_ADD_HEAD and something else in our build.
Those should be functions, or at least named with prefix MG_ etc?

kentohub, 2022-02-25:
Found why the outgoing client connections at id 2,3,4 never show up
in the loop in mg_iotest. I found that the "is_closing" flag was
being set on client connections way too early in our application, which
messed up the state machine.

mg_iotest has a line in the loop "if (c->is_closing ...) continue;" which was
skipping the loop all the time.
So I looked for who was setting this flag, and it turned out to be my timer (user) event.
I disabled that and things started working better. I don't understand what's wrong with it
yet, it should have been a straightforward conversion, but it was definitely causing
problems.
So thanks for your time, I'll close this thread.

closed 2022-02-25 by kentohub