cesanta/mongoose #1469: log time not localtime

lifw555, 2022-02-17:
- My goal is: log text use localtime
- My actions were: log info to log file
- My expectation was: 
- The result I saw: 
- My question is: log time not localtime

# Environment

- mongoose version:7.6
- Compiler/IDE and SDK: mingw32-64w version mingw32-8.1.0-posix-dwarf
- Target hardware/board: win32
- Connectivity chip/module: _?_

lifw555, 2022-02-17:
log txt:
2022-02-17 07:42:37 2 mongoose.c:2706:mg_mgr_fre All connections closed

my timezone is +8 , friendly format is
2022-02-17 15:42:37 2 mongoose.c:2706:mg_mgr_fre All connections closed

lifw555, 2022-02-17:
tmp modify .c file

replace line 2039.

from “struct tm tmp, *tm = gmtime_r(&t, &tmp);” to “struct tm tmp, *tm = localtime_r(&t, &tmp);”

it works.

cpq, 2022-02-17:
Pease update to the HEAD.
Mongoose now logs time as an integer, in order to avoid dependencies of time-related functions.

closed 2022-02-17 by cpq