cesanta/mongoose #1470: Error handling issues (CHECKED_RETURN) for getsockname

jameshilliard, 2022-02-17:
I'm seeing this get flagged by a static analysis tool:
https://github.com/cesanta/mongoose/blob/1b82aa02aa3d6fecfc23ed0a94c6917f139de1ad/src/sock.c#L122
```c
Calling "getsockname((SOCKET)(size_t)c->fd, __SOCKADDR_ARG({.__sockaddr__ = &[usa.sa](http://usa.sa/)}), &slen)" without checking return value. This library function may fail and return an error code.
```

cpq, 2022-02-19:
Does your static analyzer shut up if the function call is prefixed with `(void)` , like this:

```c
  (void) getsockname(FD(c), &usa.sa, &slen);
```

jameshilliard, 2022-02-19:
I don't think that would make it(coverity) stop reporting unchecked returns from what I read, you can see the full report [here](https://groups.google.com/g/swupdate/c/CTxwyaoJbmg/m/CyE0GM5nAQAJ) which was run against the 7.6 amalgamation as part of the [swupdate](https://sbabic.github.io/swupdate/swupdate.html) project(note that line offsets might be slightly off due to adding SPDX identifiers to the header but the amalgamation is otherwise unmodified).

cpq, 2022-02-22:
Thank you @jameshilliard 
Does this shut up the analyzer?

```c
  if (getsockname(FD(c), &usa.sa, &slen) < 0) (void) 0;  // Ignore return code, calm static analyzers
```

jameshilliard, 2022-02-22:
> Does this shut up the analyzer?

Looks like that probably would.

commit dd08231a (2022-02-23): Fix #1470 - calm static analyzer
closed 2022-02-23 by cpq