cesanta/mongoose #1486: HTTP client "POST" request

jvo203, 2022-03-07:
- My goal is: successfully submit "POST" using the HTTP client functionality
- My actions were: using "mg_printf" to submit the HTTP headers and "mg_send" to append the "POST" buffer
- My expectation was: the body received on the server would contain data (body size > 0)
- The result I saw: 

110dba9b6 3 mongoose.c:3551:accept_conn 8 accepted 133.40.215.75:57458
110dba9b6 2 ws.c:37:mg_http_ws_callback New request to: [/progress/ALMA01059729], body size: 0
<progress> POST request for 'ALMA01059729': progress = 0
110dba9b6 1 mongoose.c:400:mg_error     8 HTTP parse:

- My question is: Has anything changed between early January and March around mg_printf and mg_send? Using the mongoose version from early January (that did not contain "mg_http_status()") I can successfully submit the 4 bytes via "POST" as per the code below. But with the latest mongoose the received body size is always 0, and there is also a strange unknown "HTTP parse:" error. Am I doing anything wrong? With the exception of the previously unavailable "mg_http_status()" function, this could used to work fine.
- My code:
```
static void progress_fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data)
{
    progress_t *req = (progress_t *)fn_data;

    if (ev == MG_EV_CONNECT)
    {
        // Connected to server. Extract host name from URL
        struct mg_str host = mg_url_host(req->url);

        // Send request
        mg_printf(c,
                  "POST %s HTTP/1.0\r\n"
                  "Host: %.*s:%d\r\n"
                  "Content-Type: application/octet-stream\r\n"
                  "Content-Length: %zu\r\n"
                  "\r\n",
                  mg_url_uri(req->url),
                  (int)host.len, host.ptr, req->port, sizeof(int));

        // append the "POST" data buffer
        mg_send(c, &(req->progress), sizeof(int));
    }
    else if (ev == MG_EV_HTTP_MSG)
    {
        // Response is received. Print it
        struct mg_http_message *hm = (struct mg_http_message *)ev_data;
        printf("%.*s", (int)hm->message.len, hm->message.ptr);

        // Examine the HTTP response code. Upon success ('OK') set the return value <counter> to <progress>
        if (mg_http_status(hm) == 200)
            *(req->counter) = req->progress;

        c->is_closing = 1; // Tell mongoose to close this connection
        req->done = true;  // Tell event loop to stop
    }
    else if (ev == MG_EV_ERROR)
    {
        req->done = true; // Error, tell event loop to stop
    }
}
```

# Environment

- mongoose version: 7.6
- Compiler/IDE and SDK: gcc (Clear Linux OS for Intel Architecture) 11.2.1 20220302 releases/gcc-11.2.0-833-g8eb36ce557
- Target hardware/board: 11th Gen Intel(R) Core(TM) i7-11700 @ 2.50GHz
- Connectivity chip/module: N/A

lixiaogang5, 2022-03-07:
Hi, is this version 7.6? I can't find a declaration for the function mg_http_satatus.

jvo203, 2022-03-07:
Yes, definitely 7.6. See https://github.com/cesanta/mongoose/blob/master/mongoose.h

```#define MG_VERSION "7.6"```

and ```int mg_http_status(const struct mg_http_message *hm);```

lixiaogang5, 2022-03-07:
This function is new, [](https://github.com/cesanta/mongoose/commit/63261ec0d5b13c88d3afc86a0c4e714da337ce2c). It has not been released yet

lixiaogang5, 2022-03-07:
This function is new and has not been released yet. [mg_http_status](https://github.com/cesanta/mongoose/commit/63261ec0d5b13c88d3afc86a0c4e714da337ce2c)

jvo203, 2022-03-07:
Do you think the new "mg_http_status" function might be causing some unexpected problems? Some HTTP parse errors etc.?

Anyway, the "mg_http_status" function is called later in the code, after already having received the zero-body-sized HTTP POST message.

lixiaogang5, 2022-03-07:
I'm not sure. Since I am using the release version, the environment may be different. In addition, I do not have the implementation code of your customized callback function on the Mongoose server side, so I cannot make a specific analysis.

jvo203, 2022-03-08:
Am closing this issue for now. As expected reverting to the last stable release 7.6 from the 21st January fixes all the issues / errors (no transmission and / or parsing errors). The only "snag" is that there is no "mg_http_status()" in 7.6 yet. Using "atoi(hm->uri.ptr)" instead works fine.

I guess the current master branch is a bit unstable, it seems to be in active development. As per my original post, something must have changed since the middle of January.

lixiaogang5, 2022-03-08:
I'm sorry to see your reply so late. We'll test it again after the latest release.

jvo203, 2022-03-08:
OK, the mongoose server side is not complicated, mainly

```
static void mg_http_ws_callback(struct mg_connection *c, int ev, void *ev_data, void *fn_data)
{
// (...)
case MG_EV_HTTP_MSG:
    {
        struct mg_http_message *hm = (struct mg_http_message *)ev_data;

        MG_INFO(("New request to: [%.*s], body size: %lu", (int)hm->uri.len, hm->uri.ptr, (unsigned long)hm->body.len));
}
}
```

In the above MG_INFO call ```hm->body.len``` shows "0" when using the unstable master tree, and the correct "4" when using the stable 7.6 release. In addition, in my simple tests using only a single call to ```mg_printf(c, ...)``` succeeds, i.e.

```
// Send request
        mg_printf(c,
                  "POST %s HTTP/1.0\r\n"
                  "Host: %.*s:%d\r\n"
                  "Content-Type: application/octet-stream\r\n"
                  "Content-Length: %zu\r\n"
                  "\r\n"
                  "0000",
                  mg_url_uri(req->url),
                  (int)host.len, host.ptr, req->port, sizeof(int));
```

(notice the extra ```"0000"```) the body size 4 is transmitted and received by the master tree mongoose. However, when the "POST" body buffer is appended using ```mg_send(c, )``` after sending the headers with ```mg_printf(c, ...)```, the current master tree mongoose fails.

Using such a sequence of two calls: ```mg_printf(c)``` for the header and ```mg_send(c)``` for the body works fine in the stable 7.6 version as of the 21st January.

lixiaogang5, 2022-03-08:
The mg_printf function internally calls mg_send. These two functions separately fire two read events for select. For HTTP POST requests, there must be an entity part, otherwise http_CB will fail to parse, which is why you can just add entity "0000" here. The way you used it before was not correct. If that's right, mongoose has some internal flaws.

There are two ways to get the results you want:
(1) The mg_printf function also fills in the solid part, as you used above；
(2) const char *str = "POST %s HTTP/1.0\r\nHost: %.*s:%d\r\nContent-Type: application/octet-stream\r\n\Content-Length: %zu\r\n\r\n\"0000\"";
   mg_send(c, str, strlen(str));

jvo203, 2022-03-08:
I see ... Only one call should be used (either mg_send or mg_printf). The problem is that the POST data is not text, it is a binary buffer containing four bytes (a binary dump of <int32>). The printf format specifier has nothing like %b for bytes etc. And a naive ```"%.*s",``` only works correctly with strings, not with a binary buffer (tried and failed).

Am looking for an elegant method to first format/write the text HTTP POST header part into a temporary binary buffer, then append the binary representation of the int32 number to it and use one ```mg_send()``` to transmit it.

Frankly it would have been much easier had the new mongoose accepted the two separate calls: first send the headers with ```mg_printf``` down the Ethernet wire and then send the actual binary buffer containing the body part using ```mg_send```, just as the stable 7.6 version does at the moment (whether it is an unintended mistake on part mongoose I don't know, I just know it works right now in 7.6!).

Creating / merging two temporary buffers (text headers + binary POST body part) does not seem elegant and it wastes CPU cycles / source code lines ... Do-able but a hack.

cpq, 2022-03-15:
@jvo203 I read all messages in this issue but still have no clue - what is the problem.

Can you create a failing unit test that demonstrates an issue?

jvo203, 2022-03-16:
To all: the issue can be closed. Using the latest mongoose.{h,c} from the master branch the binary POST data is transmitted without problems in my software under development.

For completeness here is an expected correct output from simple client/server test programs. The expectation was that all 16 floating-point values would be received on the server side. And yes, all the test values have been received.

# expected output

```
chris@zodiac mongoose % ./server 
Starting WS listener on http://0.0.0.0:8000
b7efec 3 mongoose.c:2797:mg_listen      1 http://0.0.0.0:8000 port 0
b80e7c 3 mongoose.c:3545:accept_conn    2 accepted 127.0.0.1:49952
received <16> floating-point numbers.
1.000000 2.000000 3.000000 4.000000 5.000000 6.000000 7.000000 8.000000 9.000000 10.000000 11.000000 12.000000 13.000000 14.000000 15.000000 16.000000 
b80e7c 3 mongoose.c:3271:iolog            2 00100000000010 0:0 -1 err 0 (Undefined error: 0)
b80e7c 3 mongoose.c:2753:mg_close_conn  2 closed
^C
```

The source code for client and server that works OK with both mongoose 7.6 from January and the latest master branch:

# client

```
// Copyright (c) 2021 Cesanta Software Limited
// All rights reserved
//
// Example HTTP client. Connect to `s_url`, send request, wait for a response,
// print the response and exit.
// You can change `s_url` from the command line by executing: ./example YOUR_URL
//
// To enable SSL/TLS for this client, build it like this:
//    make MBEDTLS=/path/to/your/mbedtls/installation
//    make OPENSSL=/path/to/your/openssl/installation

#include "mongoose.h"

// The very first web page in history. You can replace it from command line
static const char *s_url = "http://localhost:8000/float";
// static const char *s_post = NULL; // POST data

#define N 16

float post_buf[N];

// Print HTTP response and signal that we're done
static void fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data)
{
    if (ev == MG_EV_CONNECT)
    {
        // Connected to server. Extract host name from URL
        struct mg_str host = mg_url_host(s_url);

        // Send request
        int content_length = sizeof(post_buf);
        mg_printf(c,
                  "%s %s HTTP/1.0\r\n"
                  "Host: %.*s:8000\r\n"
                  "Content-Type: octet-stream\r\n"
                  "Content-Length: %d\r\n"
                  "\r\n",
                  "POST", mg_url_uri(s_url), (int)host.len,
                  host.ptr, content_length);

        mg_send(c, post_buf, content_length);
    }
    else if (ev == MG_EV_HTTP_MSG)
    {
        // Response is received. Print it
        struct mg_http_message *hm = (struct mg_http_message *)ev_data;
        printf("%.*s", (int)hm->message.len, hm->message.ptr);
        c->is_closing = 1;       // Tell mongoose to close this connection
        *(bool *)fn_data = true; // Tell event loop to stop
    }
    else if (ev == MG_EV_ERROR)
    {
        *(bool *)fn_data = true; // Error, tell event loop to stop
    }
}

int main(int argc, char *argv[])
{
    struct mg_mgr mgr; // Event manager
    bool done = false; // Event handler flips it to true
    int i;

    // fill-in the floating-point POST buffer
    for (i = 0; i < N; i++)
        post_buf[i] = 1 + i;

    mg_log_set("3");                         // Set to 0 to disable debug
    mg_mgr_init(&mgr);                       // Initialise event manager
    mg_http_connect(&mgr, s_url, fn, &done); // Create client connection
    while (!done)
        mg_mgr_poll(&mgr, 1000); // Infinite event loop
    mg_mgr_free(&mgr);           // Free resources
    return 0;
}
```

# server

```
// Copyright (c) 2020 Cesanta Software Limited
// All rights reserved

#include <signal.h>
#include "mongoose.h"

static const char *s_url = "http://0.0.0.0:8000";

// Event handler for the listening connection.
static void cb(struct mg_connection *c, int ev, void *ev_data, void *fn_data)
{
    if (ev == MG_EV_HTTP_MSG)
    {
        struct mg_http_message *hm = (struct mg_http_message *)ev_data;

        // MG_INFO(("New request to: [%.*s], body size: %lu", (int)hm->uri.len, hm->uri.ptr, (unsigned long)hm->body.len));

        if (mg_strstr(hm->uri, mg_str("/float")) != NULL)
        {
            int i, n;

            n = hm->body.len / sizeof(float);

            printf("received <%d> floating-point numbers.\n", n);

            if (n > 0)
            {
                float *numbers = (float *)hm->body.ptr;

                for (i = 0; i < n; i++)
                    printf("%f ", numbers[i]);
                printf("\n");
            }
        }

        mg_http_reply(c, 200, NULL, "OK");
    }

    (void)fn_data;
}

int main(int argc, char *argv[])
{
    struct mg_mgr mgr; // Event manager
    mg_mgr_init(&mgr); // Initialise event manager
    mg_log_set("3");
    printf("Starting WS listener on %s\n", s_url);

    mg_http_listen(&mgr, s_url, cb, NULL); // Create HTTP listener

    while (true)
        mg_mgr_poll(&mgr, 1000); // Infinite event loop

    mg_mgr_free(&mgr);
}
```

closed 2022-03-16 by jvo203