cesanta/mongoose #1497: Passing NULL to mg_ws_connect results in malformed websocket upgrade header due to non-zeroed buffer

PythagoRascal, 2022-03-20:
- My goal is: I'm trying to learn how to use mongoose for a websocket client. I want to compile the [websocket-client tutorial](https://github.com/cesanta/mongoose/blob/2c50069/examples/websocket-client/main.c) and get consistent results.
- My actions were: I compiled the websocket-client tutorial and tested it using [wscat](https://github.com/websockets/wscat). It worked very inconsistently and I kept getting the error `mongoose\mongoose.c:4992:mg_w 1 WS handshake error: HTTP/1.1 400 Ba` (yes, it cuts off).
- My expectation was: That the tutorial code can connect to wscat consistently and not only occasionally.
- The result I saw:

Through debugging `mg_ws_connect` it seems to happen because `mem1` stays uninitialised/not zeroed when the `fmt` argument is `NULL`. The offending line seems to be this: [Link](https://github.com/cesanta/mongoose/blob/1141ea7933d1d50342a2c93404392a484f0127a0/mongoose.c#L5072). The resulting websocket upgrade header will then look something like this (notice the random characters before `Sec-WebSocket-Version`):

```
GET / HTTP/1.1
Upgrade: websocket
Host: localhost
Connection: Upgrade
á^'ìn☻Sec-WebSocket-Version: 13
Sec-WebSocket-Key: KSO+hOFs1q5SkEnx8bvp6w==

```

- My question is: Would zeroing `mem1` be enough of a fix or are there more aspects to be considered with the `%.*s` in the format string for the headers?

# Environment

- mongoose version: _7.6_
- Compiler/IDE and SDK: clang version _13.0.0_ (via `zig cc --target=x86_64-windows-gnu`, Zig version: 0.9.0-dev.1343+75cecef63)
- Target hardware/board: _Intel Z170_
- Connectivity chip/module: _?_

cpq, 2022-03-21:
@PythagoRascal maybe `n1` in the `mg_asnprintf()` call should be cast to `int` on win64.
Please pull and retry.

PythagoRascal, 2022-03-21:
Your change works now, the `\0` did the trick. Thank you!

closed 2022-03-21 by PythagoRascal