cesanta/mongoose #1501: src/sock.c:542:5: warning: conversion to 'long unsigned int' from 'int' may change the sign of the result

subnut, 2022-03-29:
- My goal is: _to `make linux-libs` on x86_64-musl (NOT glibc)_

# Environment

- mongoose version: _X7.6_
- Compiler/IDE and SDK: gcc version 10.2.1
- Target hardware/board: _x86-64_

# Elephant in the room
```
src/sock.c: In function 'mg_iotest':
src/sock.c:542:5: warning: conversion to 'long unsigned int' from 'int' may change the sign of the result [-Wsign-conversion]
  542 |     if (FD(c) > maxfd) maxfd = FD(c);
      |     ^~~~~~
src/sock.c:542:5: warning: conversion to 'long unsigned int' from 'int' may change the sign of the result [-Wsign-conversion]
src/sock.c:545:7: warning: conversion to 'long unsigned int' from 'int' may change the sign of the result [-Wsign-conversion]
  545 |   }
      |       ^     
src/sock.c:545:7: warning: conversion to 'long unsigned int' from 'int' may change the sign of the result [-Wsign-conversion]
src/sock.c:556:49: warning: conversion to 'long unsigned int' from 'int' may change the sign of the result [-Wsign-conversion]
  556 |     c->is_writable = FD(c) != INVALID_SOCKET && FD_ISSET(FD(c), &wset);
      |                                                 ^~~~~~~~
src/sock.c:556:49: warning: conversion to 'long unsigned int' from 'int' may change the sign of the result [-Wsign-conversion]
src/sock.c:557:49: warning: conversion to 'long unsigned int' from 'int' may change the sign of the result [-Wsign-conversion]
  557 |   }
      |                                                 ^       
src/sock.c:557:49: warning: conversion to 'long unsigned int' from 'int' may change the sign of the result [-Wsign-conversion]
```

subnut, 2022-03-29:
Ping @cpq

cpq, 2022-03-30:
That's actually a musl's header problem.
See https://inbox.vuxu.org/musl/20200803174707.GF6949@brightrain.aerifal.cx/T/

closed 2022-03-30 by subnut