cesanta/mongoose #1502: When websocket client issues close, close event has error code 1006

JamesSCADA98765, 2022-03-29:
I'm using Mongoose as a Web Socket server, and everything is working great. However, I am having an issue when a client closes the web socket (issues a message with option code 8). After sending the close message, the client generates an error code of 1006 and indicates the close event was not clean. Here is a copy of the error message:
CloseEvent {isTrusted: true, wasClean: false, code: 1006, reason: '', type: 'close', …}

The client is simple javascript web app running in a Microsoft Edge browser. I'm using the native WebSocket library in Javascript.

To ensure the issue was not with the client, I connected the client to a public web socket test server. When I issue a close message, the close event gives a good code of 1000 and wasClean is true. I therefore believe the issue to be with the Mongoose server rather than the client.

After doing some homework on this, it appears this error code can appear for a variety of reasons. When it occurs on close, it's generally because something is wrong with the goodby handshake. After a review of the Mongoose source code, it appears Mongoose does not automatically respond with a close acknowledgement when a close message is received from the client. Per the web socket specification, I see this is required. Is this the issue? Am I supposed to manually include mg_ws_send(connection, 0, 0, WEBSOCKET_OP_CLOSE) in the MS_EV_CLOSE event handler?

Thanks.

JamesSCADA98765, 2022-03-29:
ping @cpq

cpq, 2022-04-05:
@JamesSCADA98765 please sync and see if https://github.com/cesanta/mongoose/commit/2f77855ec14536cc62382652959c01d8cc58ca19 fixes an issue

JamesSCADA98765, 2022-04-07:
@cpq 

Thanks! That did it!

alex-chou9, 2022-09-26:
Using the lastest Mongoose v7.8, error code 1006 happens again. When any client browser (Chrome/Firefox/Safari) call web socket.close(), the browser will receive the 1006 error.

scaprile, 2022-09-26:
AFAICS there haven't been any changes in that section, can you provide a network capture so we can check what is going on ?

alex-chou9, 2022-09-27:
Upon further investigation, only Safari receive code 1006. Chrome/Firefox will receive code 1005.
network capture:
[Chrome](https://imgur.com/a/Xd6LSuB)
[Safari](https://imgur.com/a/8Ziqgh9)

In my test, if I modify the mongoose.c line 5840
`mg_ws_send(c, "", 0, WEBSOCKET_OP_CLOSE);`
to something like this
`char statusCode[2] = {0x3, 0xE8};
mg_ws_send(c, (void*)statusCode, 2, WEBSOCKET_OP_CLOSE);`

then both Chrome/Firefox will receive normal closure code 1000.
But Safari still receive 1006. (Seems like Safari not getting the WebSocket's close frame from Mongoose server.)

scaprile, 2022-09-27:
Please, post capture files, not screen snapshots.

AFAICS either you are not capturing the server response or it doesn't send it when the client is Safari, which is weird, but I can't see both requests on those snapshots.

As per [the RFC](https://www.rfc-editor.org/rfc/rfc6455):
>   1005 is a reserved value and MUST NOT be set as a status code in a
      Close control frame by an endpoint.  It is designated for use in
      applications expecting a status code to indicate that no status
      code was actually present.

That is correct, Mongoose is not sending a status code. What you do is send a status code 1000 so your software receives a status code.

As per the RFC:
> The Close frame MAY contain a body (the "Application data" portion of
   the frame) that indicates a reason for closing

"MAY" means it is not mandatory.
1005 is not an error, it is merely an indication that a status code has not been provided.
However, 1006 is:
>    1006 is a reserved value and MUST NOT be set as a status code in a
      Close control frame by an endpoint.  It is designated for use in
      applications expecting a status code to indicate that the
      connection was closed abnormally, e.g., without sending or
      receiving a Close control frame.

As you report, even though you are providing a status code your browser seems to be closing the connection (I need to see TCP to be sure of that).

Piece of advice: 
> Close Code 1006 is a special code that means the connection was closed abnormally (locally) by the browser implementation.
If your browser client reports close code 1006, then you should be looking at the websocket.onerror(evt) event for details.

We can send a status code (as you did) but that doesn't seem to solve your problem.

alex-chou9, 2022-09-28:
Network capture files: (can be opened via Wireshark)
[Safari](https://www.dropbox.com/s/ifbaxtjpp67z3lo/safari.pcapng?dl=0)
[Chrome, for reference](https://www.dropbox.com/s/i4mpkcqp2sz0dsj/chrome.pcapng?dl=0)

Both files are generated using the same, unmodified Mongoose server. And the same vanilla javascript native WebSocket API.

In Safari, websocket.onerror(evt), the evt is just a generic Event object, the object has a property "type", and its value is a string "Error".
In Safari, websocket.onclose(evt), the evt is a CloseEvent object, the object has a property "code", its value is 1006. It also has a property "reason", its value is an empty string. 
From the client side, can't seem to get useful information.

scaprile, 2022-09-28:
Thanks.
As told, there were no changes in that department on Mongoose. Looks like newer versions of Safari close the connection immediately after sending the WebSocket close command. Frame 8 is the WebSocket close command and frame 9 is the TCP close from the browser side.
Chrome, on the other side, waits for the server to close, that allows the server to send its close command and close the TCP connection.
Check on Safari documentation and best practices how you are supposed to deal with this, maybe there is some way to tell Safari not to close immediately but to wait for the server to close (as that is what it looks like you are trying to achieve); we'll check the RFC and see if there is something we could do on our side.

cpq, 2022-10-10:
@alex-chou9 can you come up with a minimal web page (with JS WS code) that illustrates the problem, in order for us to reproduce on our side?

alex-chou9, 2022-10-11:
@cpq 

index.html

```
<!DOCTYPE html>
<html>
<head>
    <script>
        window.onload = function() {
            let ws = new WebSocket("ws://192.168.1.138:8000/websocket");
            ws.onopen = (event) => {
                console.log('WebSocket open');
                setTimeout(function(){
                    ws.close();
                },5000)
                };
            ws.onclose = (event) => {
                console.log('WebSocket close. wasClean = ' + event.wasClean + ', code = ' + event.code);
            };
        };
    </script>
</head>

<body></body>
</html>
```

cpq, 2022-10-17:
@alex-chou9 the issue is with Safari.
The same error you'll see if you use other WS servers, for example, node-based.
Closing this.

closed 2022-10-17 by cpq