cesanta/mongoose #1503: ESP32 File Serving issue

matti122, 2022-03-30:
I want to serve a Javascript application from SPIFFS Partition
But I get the following error message when I request the Application from a Client.
The code is almost the same as in ESP32 example.

64be 3 mongoose.c:3517:accept_conn      2 accepted 192.168.30.58:58754
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0xb33fffff  PS      : 0x00060c30  A0      : 0x8017f85d  A1      : 0x3ffc8bf0
A2      : 0x3fff18e0  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3fff1694
A6      : 0x3ffc8cc0  A7      : 0x00000037  A8      : 0x800de347  A9      : 0x00000000
A10     : 0x3ffaf2d8  A11     : 0x3fff36dc  A12     : 0x00001000  A13     : 0x3ffc8b80
A14     : 0x3ffc8b84  A15     : 0x3ffc8b88  SAR     : 0x00000008  EXCCAUSE: 0x00000014
EXCVADDR: 0xb33ffffc  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace:0x733ffffc:0x3ffc8bf00x4017f85a:0x3ffc8c10 0x400de6d5:0x3ffc8c30 0x400de9fd:0x3ffc8d20 0x400df9f1:0x3ffc8d40 0x400dfe49:0x3ffc8d80 0x400dabcf:0x3ffc8dd0 0x400d851f:0x3ffc8df0 0x40184b81:0x3ffc8e10 0x4008f7d1:0x3ffc8e30
0x4017f85a: mg_call at *\build/../main/webconfig/../../components/mongoose/mongoose.c:396 (discriminator 1)

0x400de6d5: iolog(mg_connection*, char*, long, bool) at *build/../main/webconfig/../../components/mongoose/mongoose.c:3272

0x400de9fd: write_conn(mg_connection*) at *\build/../main/webconfig/../../components/mongoose/mongoose.c:3414 (discriminator 4)

0x400df9f1: mg_mgr_poll at *\build/../main/webconfig/../../components/mongoose/mongoose.c:3695 (discriminator 1)

0x400dfe49: WebconfigServer::initialize() at *\build/../main/webconfig/webconfig-server.cpp:105 (discriminator 1)

0x400dabcf: start_webconfig_STA at *\build/../main/webconfig/webconfig-main.cpp:79

0x400d851f: app_main at *\build/../main/main.cpp:79 (discriminator 8)

0x40184b81: main_task at F:/Espressif/esp-idf/components/freertos/port/port_common.c:129 (discriminator 2)

0x4008f7d1: vPortTaskWrapper at F:/Espressif/esp-idf/components/freertos/port/xtensa/port.c:131
# Environment

- mongoose version: latest master
- Compiler/IDE and SDK: ESP-IDF 4.4
- Target hardware/board: Custom Board
- Connectivity chip/module: ESP32-Wrover

Could you please help
Thanks

phagemann, 2022-04-02:
Same here. 
Applies to master (1c2d67de) and 7.6. 
Reverting to 7.5 solved this issue for me.

matti122, 2022-04-04:
Hello,
are there any plans to fix this issue or is ESP32 deprecated?
Thanks for your help

cpq, 2022-04-04:
You're giving a backtrace for your own application, not for a stock example.
A stock example does not crash.

matti122, 2022-04-04:
Hello cpq,
I took the stock example for ESP32 and there is this the same error.
45c 3 net.c:203:mg_listen               1 0x36 http://0.0.0.0:80
460 2 main.c:55:app_main                Starting Mongoose web server v7.6
1a05 3 sock.c:373:accept_conn           2 accepted ***
1a08 3 sock.c:257:read_conn             2 0x37 0:0 354 err 88 (Function not implemented)
1a10 3 sock.c:268:write_conn            2 0x37 1561:0 1561 err 2 (No such file or directory)
1a4b 3 sock.c:257:read_conn             2 0x37 0:0 307 err 2 (No such file or directory)
1a54 3 sock.c:268:write_conn            2 0x37 56:0 56 err 2 (No such file or directory)

This error also appears with latest master in my custom implementation.
I'm build the APP on a Windows host.
Could you please check.
Thanks

cpq, 2022-04-05:
Yes there was a problem with the example code.
Please fetch the latest version and retry.

matti122, 2022-04-05:
Hello cpq,
sorry but it still not working. One time a file will be delivered but if switch to another site the Server is crashing.
Maybe you can check again please.
Thanks
Here is the Log.
W (69575) wifi:<ba-add>idx:2 (ifx:1, 42:98:8a:51:dd:3e), tid:0, ssn:7, winSize:64
W (69625) wifi:<ba-add>idx:3 (ifx:1, 42:98:8a:51:dd:3e), tid:1, ssn:0, winSize:64
W (70225) wifi:<ba-del>idx
W (70225) wifi:<ba-add>idx:4 (ifx:1, 42:98:8a:51:dd:3e), tid:5, ssn:0, winSize:64
W (70225) wifi:<ba-del>idx
W (70225) wifi:<ba-add>idx:2 (ifx:1, 42:98:8a:51:dd:3e), tid:0, ssn:12, winSize:64
W (70235) wifi:<ba-add>idx:3 (ifx:1, 42:98:8a:51:dd:3e), tid:1, ssn:1, winSize:64
W (70595) wifi:<ba-del>idx
W (70595) wifi:<ba-del>idx
W (70595) wifi:<ba-add>idx:2 (ifx:1, 42:98:8a:51:dd:3e), tid:0, ssn:15, winSize:64
E (70595) wifi:CCMP replay detected: A1=a8:03:2a:e2:0f:75 A2=42:98:8a:51:dd:3e PN=36, RSC=37 seq=63232
W (70605) wifi:<ba-add>idx:4 (ifx:1, 42:98:8a:51:dd:3e), tid:5, ssn:1, winSize:64
W (71915) wifi:<ba-del>idx
W (71915) wifi:<ba-del>idx
W (71915) wifi:<ba-add>idx:2 (ifx:1, 42:98:8a:51:dd:3e), tid:0, ssn:21, winSize:64
E (71915) wifi:CCMP replay detected: A1=a8:03:2a:e2:0f:75 A2=42:98:8a:51:dd:3e PN=46, RSC=47 seq=63696
W (71925) wifi:<ba-add>idx:4 (ifx:1, 42:98:8a:51:dd:3e), tid:5, ssn:1, winSize:64
W (72155) wifi:<ba-del>idx
E (72155) wifi:CCMP replay detected: A1=a8:03:2a:e2:0f:75 A2=42:98:8a:51:dd:3e PN=51, RSC=52 seq=63792
W (72155) wifi:<ba-add>idx:4 (ifx:1, 42:98:8a:51:dd:3e), tid:5, ssn:2, winSize:64
10fee 3 mongoose.c:3389:read_conn       2 0x37 0:0 338 err 88 (Function not implemented)
110be 3 mongoose.c:3389:read_conn       5 0x3a 0:0 338 err 88 (Function not implemented)
W (72585) wifi:<ba-del>idx
W (72585) wifi:<ba-add>idx:4 (ifx:1, 42:98:8a:51:dd:3e), tid:5, ssn:2, winSize:64
1119c 3 mongoose.c:3389:read_conn       4 0x39 0:0 355 err 88 (Function not implemented)
11260 3 mongoose.c:3389:read_conn       3 0x38 0:0 355 err 88 (Function not implemented)
1126f 3 mongoose.c:3400:write_conn      2 0x37 4096:0 4096 err 88 (Function not implemented)
1127f 3 mongoose.c:3400:write_conn      5 0x3a 4096:0 4096 err 88 (Function not implemented)
11288 3 mongoose.c:3400:write_conn      4 0x39 908:0 908 err 88 (Function not implemented)
11293 3 mongoose.c:3400:write_conn      3 0x38 4096:0 4096 err 88 (Function not implemented)
11299 3 mongoose.c:3400:write_conn      2 0x37 4096:0 0 err 11 (No more processes)
1129b 3 mongoose.c:3400:write_conn      5 0x3a 4096:0 0 err 11 (No more processes)
114e9 3 mongoose.c:3400:write_conn      5 0x3a 4096:0 0 err 11 (No more processes)
114ea 3 mongoose.c:3400:write_conn      2 0x37 4096:0 0 err 11 (No more processes)
118d1 3 mongoose.c:3400:write_conn      5 0x3a 4096:0 0 err 11 (No more processes)
118d2 3 mongoose.c:3400:write_conn      2 0x37 4096:0 0 err 11 (No more processes)

matti122, 2022-04-05:
I tested it again with a Windows Client. With Android and IOS Clients the website is not loading. It is not very stable. But also websocket server is not working on ESP32 anymore.

cpq, 2022-04-06:
That log shows a low-level wifi issue , not related to mongoose.
A more complete log would be nice.

closed 2022-04-13 by cpq