cesanta/mongoose #1511: Usability issues with mg_mkpipe()

sebhub, 2022-04-05:
I would like to use a separate thread which sends and receives data via a websocket through a pipe. When a websocket is opened (MG_EV_WS_OPEN) I create a pipe with mg_mkpipe() and start a thread which gets the pointer to new connection object as a parameter.

I can send data to the pipe from the worker thread using mg_mgr_wakeup(). The data is received through the pipe handler MG_EV_READ  event in the manager thread.

I can send data from the manager thread to the worker thread using mg_send(). **How can I receive the data in the worker thread?**

How can I clean up the pipe if the websocket is closed? I can probably use mg_close_conn(). However, I can do this only if the worker thread doesn't use the connection structure currently (SMP system for example) and will not use it in the future. I want to avoid waiting for the worker thread. In addition, how can I close the socket used by the worker thread? One option would be to split up the connection structure into two separately allocated and closed structures: one for the manager thread and one for the worker thread.

sebhub, 2022-04-05:
Another question, why does mg_socketpair() create two UDP sockets and doesn't use socketpair(PF_UNIX, ...)?

cpq, 2022-04-05:
@sebhub 

1. You can't receive any data , only send
2. If you want to signal something back to the thread that holds a pipe, use `pipe->label`
3. A socketpair syscall is not used in order to have a single codebase on all architectures, even those that don't have that syscall

closed 2022-04-05 by cpq