cesanta/mongoose #1522: buffered TLS data produces deadlock.

david-collett, 2022-04-11:
Hi All,
I have experienced a read deadlock on mongoose master when using TLS with mongoose http client.
I am using a custom TLS for wolfSSL, but I believe the bug affects the builtin TLS support also.
The issue is as follows:

read_conn requests 2048 bytes of data.
internally, SSL reads more than this (say, 3000bytes), in order to obtain a whole tls record to decrypt.
2048 bytes is returned and passed to the callback. The rest is buffered inside wolfSSL. (OpenSSL appears to behave the same way).
As the server has no more data to send at this point, select() does not mark the connection as readable, so read_conn (and thus the tls read method) is not called again and the process stalls.

It appears mongoose used to have mitigations for this issue, but they were removed in commit 02cabcc0ecf09f47971e4c62c90afb0cf155cbef on Mar 14.

Was removing this a mistake?

cpq, 2022-04-11:
@david-collett thank you. Could you put together a quick test (a github project?) that reproduces an issue?

david-collett, 2022-04-11:
Sure, I'll try.

david-collett, 2022-04-11:
Please see the very minor changes to an example to reproduce the issue in my forked branch here:

https://github.com/david-collett/mongoose/tree/tls-buffer-stalemate

The instructions (also in the commit message) are:

    Steps to reproduce

    cd examples/http-streaming-client
    OPENSSL=1 make example
    
    ./example https://httpbin.org/bytes/2000

    The bug is reproduced when:
    - The server reply body is in a tls record that is larger than
      the initial mongoose buffer size (2048) such that the first
      read_conn results in buffering in the tls layer.
    - The server does not send anything else (so select() does not
      mark the socket)
    - The server keeps the connection alive.
    
    The test URL above meets these criteria. You may have to run it
    more than once to produce the failure.

Reverting 02cabcc0ecf09f47971e4c62c90afb0cf155cbef fixes it.

Let me know if you need anything more.

Dave

cpq, 2022-04-12:
Can't reproduce. Cloned your repo, running example against suggested site, and others. It does not get stuck.
Using Ubuntu 20.4, mbedtls 2.16, openssl 2.1.2

david-collett, 2022-04-13:
ok, that's frustrating, were you definitely using my tls-buffer-stalemate branch?
I am on Ubuntu 21.10 with openssl 1.1.1l.

This is what a failed run looks like:
```
dave@morty:~/src/mongoose/examples/http-streaming-client$ ./example https://httpbin.org/bytes/2000
d592b2f 3 net.c:181:mg_connect          1 0xffffffffffffffff https://httpbin.org/bytes/2000
d592b2f 3 net.c:181:mg_connect          2 0xffffffffffffffff udp://8.8.8.8:53
d592b2f 3 sock.c:355:mg_connect_resolve 2 0x3
d592b2f 3 sock.c:151:mg_send            2 0x3 0:0 29 err 0 (Success)
d592b3a 3 sock.c:266:read_conn          2 0x3 0:0 157 err 0 (Success)
d592b3a 3 dns.c:166:dns_cb              1 httpbin.org resolved to 35.169.55.235
d592b3a 3 sock.c:355:mg_connect_resolve 1 0x4
d592b3a 2 main.c:57:fn                  POLL...
d592c25 2 main.c:57:fn                  POLL...
d592c26 3 tls_openssl.c:38:mg_tls_init  1 Setting TLS, CA: ca.pem, cert: null, key: null
d592c28 3 tls_openssl.c:115:mg_tls_init 1 SSL client OK
d592d15 2 main.c:57:fn                  POLL...
d592d18 2 main.c:57:fn                  POLL...
d592e03 2 main.c:57:fn                  POLL...
d592e03 3 tls_openssl.c:128:mg_tls_hand 1 success
d592e03 2 main.c:57:fn                  POLL...
d592e03 3 sock.c:277:write_conn         1 0x4 95:0 95 err 0 (Success)
d592ef1 2 main.c:57:fn                  POLL...
d592ef1 3 tls_openssl.c:148:mg_tls_recv Read (2048) returned -1, pending: 0
d592ef1 3 sock.c:266:read_conn          1 0x4 0:0 0 err 11 (Resource temporarily unavailable)
d592ef5 2 main.c:57:fn                  POLL...
d592ef5 3 tls_openssl.c:148:mg_tls_recv Read (2048) returned 2048, pending: 191
d592ef5 3 sock.c:266:read_conn          1 0x4 0:0 2048 err 0 (Success)
d592ef5 2 main.c:43:fn                  CHUNK 1809
d5932de 2 main.c:57:fn                  POLL...
d5932df 2 main.c:62:fn                  SSL_pending has 191 bytes!
d5936c8 2 main.c:57:fn                  POLL...
d5936c8 2 main.c:62:fn                  SSL_pending has 191 bytes!
d593ab2 2 main.c:57:fn                  POLL...
d593ab2 2 main.c:62:fn                  SSL_pending has 191 bytes!
^C
```

The issue should be clear. There is no mitigation in place for this problem, but there used to be in the form of this line in mg_mgr_poll that was removed in 02cabcc0ecf09f47971e4c62c90afb0cf155cbef.

```
while (c->is_tls && read_conn(c) > 0) (void) 0;  // Read buffered TLS data
```

Given that mongoose sockets are inherently non-blocking, I'm not sure I'd consider this a busy loop. But if so, one could possibly extent the tls abstraction with a method to only return buffered data so the above does not actually result in a read().

One thought, this problem will not occur if the server closes the socket after serving the request because select() will flag it for read in that case, hence my test adds keep-alive headers to try and induce this condition. Does mongoose support http pipelining? i.e. can you send a new request on an existing connection and still use the http response callbacks? if not, then this is not a problem in practice for mongoose http client users. However, I can see how it could affect other protocols, at least in theory.

Thanks,
Dave

cpq, 2022-04-13:
Ah, I've cloned your repo but did not use your branch. Reproduced, thank you. Looking into it.

Note - the while loop that calls read_conn() actually deadlocks on some embedded systems. It should be another way.

cpq, 2022-04-13:
@david-collett please sync and re-test

david-collett, 2022-04-13:
yep, that works great thanks.
I see that you also reset the struct tv if data is pending so there is no unnecessary sleep, looks great to me.

Thanks heaps!
Dave

cpq, 2022-04-13:
@david-collett thank you for reporting and your efforts, appreciated.