cesanta/mongoose #1529: HTTP Requests result in IO Errors from zero-length recv

fentonscode, 2022-04-18:
**Goal**

To interact with HTTP requests, more simply, to run the tutorials [here](https://mongoose.ws/tutorials/http-server/) without IO Errors littering the console.

**Actions**

- Copied the c source to the project build directory
- Added include dir to reference the main header
- Linked against `ws2_32.lib` (Separate issue: this is **not** documented anywhere as a requirement and perhaps should be noted)
- Copy/Paste tutorial code (also tested with my own code with the same results)

**My expectation**

Zero-length reads not flagged as an error on HTTP requests.

**Results**

On _every_ HTTP request, regardless of end-point, content, or method (or anything really) messages similar to the following:

`5d777fd1 1 mongoose.c:403:mg_error      4 0xe0 IO error` 

appear in the console. For clients that send a lot of requests, the log/console is quickly overwhelmed with errors (although it appears 1:1 with requests). In terms of server functionality, the HTTP request can be seen, parsed and `mg_http_reply()` responds correctly when checked with multiple REST test clients, but the console message is still printed. It appears the result of this is that when `mg_sock_recv` is called, the result is -1 (possibly indicating no data, as the HTTP request has been dealt with). The zero-length read then propagates to `mg_error` which writes the console message (there is a variable `n` that goes to -1 and triggers the condition)

**Questions**

Is there special processing required which is not presented in the documentation? For example, in the User Guide there is an example on handling the `MG_EV_READ` event that could potentially help (I could force a connection to close?). I have seen some servers close the socket immediately in the message handler, but this is not in the minimal example, so is it handled be default?

It appears that I may be using the API incorrectly, but have tried to adhere to the documented examples to ensure that it's not something I've added that's causing problems.

# Environment

- mongoose version: _7.6_
- Compiler/IDE and SDK: Visual Studio 2019 version _16.11.8_

cpq, 2022-04-19:
Fixed, please confirm.
W.r.t. `Is there special processing ...` question - could you elaborate please? I did not get the question.

fentonscode, 2022-04-19:
Thanks for looking into this. To clarify what I meant, the changes in the HTTP protocol from 0.9 to 1.1 appear to imply different behaviours of servers with respect to automatically closing connections (e.g. 1.0 requires an explicit close server-side from what I can gather - relevant summary [here](https://stackoverflow.com/questions/17437950/when-does-an-http-1-0-server-close-the-connection)). I was wondering if mongoose had something built in and I was accidentally bypassing it or suppressing it (although I did try with 1.0 and 1.1 protocol requests and had the same problem).

Specifically, by 'special processing' it was a bit unclear whether on 1.0 protocol requests (or similar that require a server-side close) needed to capture the `MG_EV_READ` event after the socket buffer was read and explicitly close to prevent the IO error in the first place.

cpq, 2022-04-19:
@fentonscode thanks.

`MG_EV_READ` is a low-level event. When it comes to HTTP, application developer should not care about it. `MG_EV_HTTP_MSG` is the event to handle, taking care about https://mongoose.ws/tutorials/http-server/#things-to-remember

You're correct - HTTP/1.0 implicitly closes the connection, whereas HTTP/1.1 keeps it alive. Mongoose does not really pays attention to the protocol version - although it assumes 1.1 behaviour, e.g. the possibility of pipelined requests. Parsed HTTP message logic is outlined at https://github.com/cesanta/mongoose/blob/master/src/http.c#L235-L246

closed 2022-04-19 by cpq