cesanta/mongoose #1530: How to install on compile on windows 10.

rajhlinux, 2022-04-18:
- My goal is: how to compile in windows 10 and able to use websockets.
- My actions were: was not able to compile.
- My expectation was: to compile 
- The result I saw: no info on compilation
- My question is: how can I compile on windows 10 or does it even requires compilation to use websocket examples?

# Environment:

windows 10

- mongoose version: todays version
- Compiler/IDE and SDK: MiniGW MSYS2 or VS 2017
- Target hardware/board: AMD 64
- Connectivity chip/module: _?_

Thanks.

rajhlinux, 2022-04-18:
I'm getting these errors:

```
C:\Users\Admin\Downloads\Compressed\mongoose-master_copy\examples\websocket-server>gcc -I C:\Users\Admin\Downloads\Compressed\mongoose-master\mongoose-master\ -W -Wall -DMG_ENABLE_LINES=1 main.c -o websocket_server
C:/TDM-GCC-64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\Admin\AppData\Local\Temp\ccvSXuCs.o:main.c:(.text+0x93): undefined reference to `mg_http_match_uri'
C:/TDM-GCC-64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\Admin\AppData\Local\Temp\ccvSXuCs.o:main.c:(.text+0xad): undefined reference to `mg_ws_upgrade'
C:/TDM-GCC-64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\Admin\AppData\Local\Temp\ccvSXuCs.o:main.c:(.text+0xc5): undefined reference to `mg_http_match_uri'
C:/TDM-GCC-64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\Admin\AppData\Local\Temp\ccvSXuCs.o:main.c:(.text+0xed): undefined reference to `mg_http_reply'
C:/TDM-GCC-64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\Admin\AppData\Local\Temp\ccvSXuCs.o:main.c:(.text+0x139): undefined reference to `mg_http_serve_dir'
C:/TDM-GCC-64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\Admin\AppData\Local\Temp\ccvSXuCs.o:main.c:(.text+0x16d): undefined reference to `mg_ws_send'
C:/TDM-GCC-64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\Admin\AppData\Local\Temp\ccvSXuCs.o:main.c:(.text+0x18d): undefined reference to `mg_mgr_init'
C:/TDM-GCC-64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\Admin\AppData\Local\Temp\ccvSXuCs.o:main.c:(.text+0x1c3): undefined reference to `mg_http_listen'
C:/TDM-GCC-64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\Admin\AppData\Local\Temp\ccvSXuCs.o:main.c:(.text+0x1d4): undefined reference to `mg_mgr_poll'
collect2.exe: error: ld returned 1 exit status
```

same goes for GCC

cpq, 2022-04-19:
Add mongoose.c to compilation

closed 2022-04-19 by cpq

rajhlinux, 2022-04-19:
Thank You for replying.

I have done so, as with the tutorial:

```
C:\Users\Admin\Downloads\Compressed\mongoose-master\mongoose-master\examples\websocket-server>clang ../../mongoose.c -I../.. -W -Wall -DMG_ENABLE_LINES=1  -o example main.c
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_socket referenced in function mg_connect_resolved
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_WSAGetLastError referenced in function mg_connect_resolved
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_connect referenced in function mg_connect_resolved
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_setsockopt referenced in function mg_tls_free
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_bind referenced in function mg_tls_free
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_listen referenced in function mg_tls_free
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_getsockname referenced in function mg_tls_free
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_closesocket referenced in function mg_tls_free
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_WSAStartup referenced in function mg_mgr_init
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_sendto referenced in function mg_sock_conn_reset
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_send referenced in function mg_sock_conn_reset
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_ioctlsocket referenced in function mg_sock_conn_reset
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_select referenced in function pf1
mongoose-295a8f.o : error LNK2019: unresolved external symbol __WSAFDIsSet referenced in function pf1
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_accept referenced in function pf1
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_getsockopt referenced in function pf1
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_recvfrom referenced in function mg_log_stdout
mongoose-295a8f.o : error LNK2019: unresolved external symbol __imp_recv referenced in function mg_log_stdout
example : fatal error LNK1120: 18 unresolved externals
clang: error: linker command failed with exit code 1120 (use -v to see invocation)
```

Similar error... I guess complier settings are messed up in windows 10 since I have installed like 5 different compilers, MVSC, GCC, MinGW, Cygwin and Clang. 

Anyhow I ran the code on FreeBSD, works fine. Would be great to get it working on Windows 10 without using MVSC.

Thanks,

cpq, 2022-04-19:
Either update to the latest HEAD, or add `ws2_32.lib` library to the build command.

rajhlinux, 2022-04-19:
Thank You for the quick reply.

Will try it.

Thanks.

rajhlinux, 2022-04-20:
Thank You it works perfectly now.

I have no idea what `ws2_32.lib` is but I did a search for it using an application called "Eveything" and it found `ws2_32.lib` on this path: `C:\Program Files (x86)\Windows Kits\10\Lib\10.0.19041.0\um\x64`

I then copied the `ws2_32.lib` file and paste it into the websocket example directory and ran the complication code on windows terminal:
`clang WS2_32.Lib ../../mongoose.c -I../.. -W -Wall -DMG_ENABLE_LINES=1 -o example.exe main.c`

An example.exe was produced and runs exactly how the mongoose websocket server tutorial explains it should.

Thanks again.

https://i.imgur.com/YWrHXlh.png