cesanta/mongoose #1532: Improve IPv6 dns support

david-collett, 2022-04-21:
In a single-stack IPv6 environment, DNS resolution can fail for a number of reasons:
- Error trying to contact the IPv4 server (if set, which it is by default).
- Failure to try the IPv6 server at all if no IPv4 server is set ("DNS server URL is NULL. Call mg_mgr_init()")

As an example, Australia's largest mobile carrier Telstra uses single-stack IPv6 for mobile devices (e.g. iphones) by default. To support communication with IPv4 hosts, they employ DNS64/NAT64. In this environment, there is no configured IPv4 DNS server. By default mongoose http client will fail in this environment. Even if you clear the mgr.dns4.url and set the mgr.dns6.url it will still fail.

To make mongoose this work in this environment, the dns6 resolver must be set to the one configured on the host (so that DNS64 is performed for IPv4 hosts), and mongoose must be forced to use dns6.

One way to do this is something like the following which will work provided mgr.dns4.url is set to NULL. A more complete fix might automatically detect failures trying IPv4 and try IPv6. I believe currently it only ever queries the ipv6 dns server if a query to the ipv4 server succeeds but fails to return an A record, this is not the failure case in a pure ipv6 environment.

```
diff --git a/mongoose.c b/mongoose.c
index 674733f6..6894b16e 100644
--- a/mongoose.c
+++ b/mongoose.c
@@ -374,7 +374,15 @@ void mg_resolve(struct mg_connection *c, const char *url) {
     mg_connect_resolved(c);
   } else {
     // host is not an IP, send DNS resolution request
-    mg_sendnsreq(c, &host, c->mgr->dnstimeout, &c->mgr->dns4, false);
+#if MG_ENABLE_IPV6
+    if (c->mgr->dns4.url == NULL) {
+      mg_sendnsreq(c, &host, c->mgr->dnstimeout, &c->mgr->dns6, true);
+    } else {
+#endif
+      mg_sendnsreq(c, &host, c->mgr->dnstimeout, &c->mgr->dns4, false);
+#if MG_ENABLE_IPV6
+    }
+#endif
   }
 }
```

# Environment

- mongoose version: master
- Compiler/IDE and SDK: Ubuntu 20.04
- Target hardware/board: _?_
- Connectivity chip/module: _?_

cpq, 2022-04-21:
Does this patch work?

```diff
diff --git a/mongoose.c b/mongoose.c
index 674733f6..0870bc34 100644
--- a/mongoose.c
+++ b/mongoose.c
@@ -279,7 +279,8 @@ static void dns_cb(struct mg_connection *c, int ev, void *ev_data,
                       mg_ntoa(&d->c->rem, buf, sizeof(buf))));
             mg_connect_resolved(d->c);
 #if MG_ENABLE_IPV6
-          } else if (dm.addr.is_ip6 == false && dm.name[0] != '\0') {
+          } else if (dm.addr.is_ip6 == false && dm.name[0] != '\0' &&
+                     c->mgr->use_dns6 == false) {
             struct mg_str x = mg_str(dm.name);
             mg_sendnsreq(d->c, &x, c->mgr->dnstimeout, &c->mgr->dns6, true);
 #endif
@@ -374,7 +375,8 @@ void mg_resolve(struct mg_connection *c, const char *url) {
     mg_connect_resolved(c);
   } else {
     // host is not an IP, send DNS resolution request
-    mg_sendnsreq(c, &host, c->mgr->dnstimeout, &c->mgr->dns4, false);
+    struct mg_dns *dns = c->mgr->use_dns6 ? &c->mgr->dns6 : &c->mgr->dns4;
+    mg_sendnsreq(c, &host, c->mgr->dnstimeout, dns, c->mgr->use_dns6);
   }
 }
 
diff --git a/mongoose.h b/mongoose.h
index d04510e8..fd45defb 100644
--- a/mongoose.h
+++ b/mongoose.h
@@ -898,6 +898,7 @@ struct mg_mgr {
   struct mg_dns dns4;           // DNS for IPv4
   struct mg_dns dns6;           // DNS for IPv6
   int dnstimeout;               // DNS resolve timeout in milliseconds
+  bool use_dns6;                // Use DNS6 server by default, see #1532
   unsigned long nextid;         // Next connection ID
   void *userdata;               // Arbitrary user data pointer
   uint16_t mqtt_id;             // MQTT IDs for pub/sub
```

david-collett, 2022-04-22:
Yep, that works, thanks.

commit a7cbc197 (2022-04-22): Fix #1532 - improve ipv6 DNS lookup
closed 2022-04-22 by cpq