cesanta/mongoose #1533: Possible bug in mg_mgr_free or mg_iotest on Windows implementations

fentonscode, 2022-04-21:
**Goal**

I have a server which is currently operating by calling `mg_mgr_poll` in a while loop controlled by a global flag, in line with the signals section of the documentation and tutorials. When this flag is changed, the server shuts down by calling `mg_mgr_free` (eventually) and the console reports a failure in `mg_iotest`.

Specifically, the call to `mg_mgr_free` calls `mg_mgr_poll` again (line 2766), which operates on a connection manager with nothing but closing sockets. When control reaches `mg_iotest` all of the parameters to `select` are zero'd out, causing a `WSAEINVAL` error code.

**Actions**

_Any_ call to `mg_mgr_free` at _any_ time appears to trigger this.

The only work-around I was able to do was to detect the `MG_EV_POLL` message, determine if all connections were closing and then throw an exception from the `fn` handler (catch outside of loop). This bypasses the `select` calls but is incredibly obtuse (and wrong).

**Expectation**

`mg_mgr_free` should be safe to call without `select` failures in most cases. Certainly on inactive servers (I.E. just a listen connection that polls a few times and then tries to clean up - as these still cause errors).

**Result**

Console prints `b72d1cb 1 mongoose.c:3594:mg_iotest     select: -1 10022` (or similar) when `mg_mgr_free` is called.

**Questions**

What is the correct way to shut down a server using this API?

# Environment

- mongoose version: _7.6_
- Compiler/IDE and SDK: Visual Studio version _16.11.12_

fentonscode, 2022-04-22:
A small update: This also consistently throws an access violation exception in `mg_resolve_cancel`, since when all connections are forced to close in `mg_mgr_free`, the `close_conn` call later in the code will always hit a null pointer.

cpq, 2022-04-22:
Please test https://github.com/cesanta/mongoose/commit/9732a7e13b1faafa6ca02294ef319a9e47b88572

closed 2022-04-22 by cpq

fentonscode, 2022-04-22:
Thanks @cpq , will test and report back later on.

fentonscode, 2022-04-25:
Issue not fixed, access violation exception still thrown when code reaches line 3660 and calls `close_conn` on a null pointer (connection appears already closed and inaccessible - `mg_resolve_cancel` throws on line 137)

commit bdd7d921 (2022-04-22): Fix #1533 - fix iotest error on windows