cesanta/mongoose #1534: when frequently fresh the websocket webpage, the server got an error

matthiashack, 2022-04-23:
[image]

[image]

matthiashack, 2022-04-23:
[test.zip](https://github.com/cesanta/mongoose/files/8546963/test.zip)

cpq, 2022-04-23:
Why do you use a non-documented, internal method that is neither in the documentation nor in the examples?

closed 2022-04-23 by cpq

matthiashack, 2022-04-24:
for example ,two clients connect to the server, one force close the connection, I need to close the pair client manally

cpq, 2022-04-24:
@matthiashack use `c->is_closing = 1` (immediate close) or `c->is_draining = 1` (send the rest of data and close)  to signal event manager to close the connection. Don't remote it from manager yourself.

cpq, 2022-04-24:
Please see https://mongoose.ws/documentation/#best-practices