cesanta/mongoose #1538: mg_tls_handshake failing on https/openssl

tomrau3366, 2022-04-28:
This is MG_VERSION 7.6

The handshake code looks like this:

void mg_tls_handshake(struct mg_connection *c) {
  struct mg_tls *tls = (struct mg_tls *) c->tls;
  int rc;
  SSL_set_fd(tls->ssl, (int) (size_t) c->fd);
  rc = c->is_client ? SSL_connect(tls->ssl) : SSL_accept(tls->ssl);
  if (rc == 1) {
    MG_DEBUG(("%lu success", c->id));
    c->is_tls_hs = 0;
  } else {
    int code;
    ERR_print_errors_fp(stderr);
    code = mg_tls_err(tls, rc);
    if (ret_ != 0) mg_error(c, "tls hs: rc %d, err %d", rc, ret_);
  }
}

rc is always returned as -1, and SSL_get_error returns SSL_ERROR_WANT_READ (2)

a7c21b 3 mongoose.c:2729:mg_connect     1 -1 https://192.168.2.119:3000
a7c21b 3 mongoose.c:3459:mg_connect_res 1 0x10c
a7c21b 4 mongoose.c:3643:mg_mgr_poll    1 -w tChrc
a7c21b 3 mongoose.c:4479:mg_tls_init    1 Setting TLS, CA: c:\mtls\certs\cacert.pem, cert: c:\client_certs\client.cert.pem, key: c:\client_certs\client.key.pem
a7ddf0 1 mongoose.c:405:mg_error        1 0x10c tls hs: rc -1, err 2
accf51 3 mongoose.c:4553:mg_tls_init    1 SSL client OK
accf51 3 mongoose.c:2706:mg_close_conn  1 closed
acd7cd 3 mongoose.c:2785:mg_mgr_free    All connections closed

I've got a simple local openssl connection running/listening on port 3000

openssl s_client -connect 192.168.2.119:3000 -key c:\client_certs\client.key.pem  -cert c:\client_certs\client.cert.pem -CAfile c:\mtls\certs\cacert.pem -state

The server side looks okay until we drop the client:

SSL_accept:before SSL initialization
SSL_accept:before SSL initialization
SSL_accept:SSLv3/TLS read client hello
SSL_accept:SSLv3/TLS write server hello
SSL_accept:SSLv3/TLS write change cipher spec
SSL_accept:TLSv1.3 write encrypted extensions
SSL_accept:SSLv3/TLS write certificate
SSL_accept:TLSv1.3 write server certificate verify
SSL_accept:SSLv3/TLS write finished
SSL_accept:TLSv1.3 early data
SSL_accept:error in TLSv1.3 early data
ERROR
shutting down SSL
CONNECTION CLOSED

The main.c is basically the same as the http_client example code:

#include "mongoose.h"

// The very first web page in history. You can replace it from command line
static const char *s_url = "https://192.168.2.119:3000";
static const char *s_post_data = NULL;      // POST data
static const uint64_t s_timeout_ms = 150000;  // Connect timeout in milliseconds

// Print HTTP response and signal that we're done
static void fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
  if (ev == MG_EV_OPEN) {
    // Connection created. Store connect expiration time in c->label
    *(uint64_t *) c->label = mg_millis() + s_timeout_ms;
  } else if (ev == MG_EV_POLL) {
    if (mg_millis() > *(uint64_t *) c->label &&
        (c->is_connecting || c->is_resolving)) {
      mg_error(c, "Connect timeout");
    }
  } else if (ev == MG_EV_CONNECT) {
    // Connected to server. Extract host name from URL
    struct mg_str host = mg_url_host(s_url);

    // If s_url is https://, tell client connection to use TLS
    if (mg_url_is_ssl(s_url)) {
      struct mg_tls_opts opts = {.ca = "c:\\mtls\\certs\\cacert.pem",
                                 .certkey = "c:\\client_certs\\client.key.pem",
                                 .cert = "c:\\client_certs\\client.cert.pem",
                                 .srvname = host};
      mg_tls_init(c, &opts);
    }

    // Send request
    int content_length = s_post_data ? strlen(s_post_data) : 0;
    mg_printf(c,
              "%s %s HTTP/1.0\r\n"
              "Host: %.*s\r\n"
              "Content-Type: octet-stream\r\n"
              "Content-Length: %d\r\n"
              "\r\n",
              s_post_data ? "POST" : "GET", mg_url_uri(s_url), (int) host.len,
              host.ptr, content_length);
    mg_send(c, s_post_data, content_length);
  } else if (ev == MG_EV_HTTP_MSG) {
    // Response is received. Print it
    struct mg_http_message *hm = (struct mg_http_message *) ev_data;
    printf("%.*s", (int) hm->message.len, hm->message.ptr);
    c->is_closing = 1;         // Tell mongoose to close this connection
    *(bool *) fn_data = true;  // Tell event loop to stop
  } else if (ev == MG_EV_ERROR) {
    *(bool *) fn_data = true;  // Error, tell event loop to stop
  }
}

int main(int argc, char *argv[]) {
  struct mg_mgr mgr;              // Event manager
  bool done = false;              // Event handler flips it to true
  if (argc > 1) s_url = argv[1];  // Use URL provided in the command line
  mg_log_set("4");                // Set to 0 to disable debug
  mg_mgr_init(&mgr);              // Initialise event manager
  mg_http_connect(&mgr, s_url, fn, &done);  // Create client connection

  while (!done) mg_mgr_poll(&mgr, 50);      // Infinite event loop
  mg_mgr_free(&mgr);                        // Free resources
  return 0;
}

I can use openssl s_client to verify proper authentication with these certs --- I'm not sure if I'm doing something wrong in the code sending these possibly?

cpq, 2022-05-01:
@tomrau3366 your code uses CA, cert and key in the TLS init.
That means, your client is not only checks the server's cert using CA, but also offers its own cert. In other words, this is a 2-way TLS, where both ends authenticate using certificates.
For that to succeed, your server should be configured accordingly. Is that your intention? Please check using `curl` utility, make sure it works, and share `curl` output.

cpq, 2022-05-13:
@tomrau3366 Any more input?

cpq, 2022-05-16:
No more feedback, closing

closed 2022-05-16 by cpq