cesanta/mongoose #1544: WebSocket client: cannot connect to wss://

jvo203, 2022-05-08:
- My goal is: open a secure WebSocket connection
- The WebSocket client source code (following the latest github master, compiled with ```-DMG_ENABLE_OPENSSL=1```):
```
#include <signal.h>
#include "mongoose.h"

static const char *s_url = "wss://socket.polygon.io/forex";

// Handle interrupts, like Ctrl-C
static int s_signo = 0;
static void signal_handler(int signo)
{
    s_signo = signo;
}

// Polygon.io forex WebSocket handler
static void fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data)
{
    if (ev == MG_EV_CONNECT)
    {
        struct mg_str host = mg_url_host(s_url);

        if (mg_url_is_ssl(s_url))
        {
            struct mg_tls_opts opts = {.ca = "ca.pem", .srvname = host};
            mg_tls_init(c, &opts);
        }
    }
    else if (ev == MG_EV_OPEN)
    {
        c->is_hexdumping = 1;
    }
    else if (ev == MG_EV_ERROR)
    {
        // On error, log error message
        MG_ERROR(("%p %s", c->fd, (char *)ev_data));
    }
    else if (ev == MG_EV_WS_OPEN)
    {
        printf("Polygon.io WebSocket connection is OPEN.\n");
        // When websocket handshake is successful, send message
        // mg_ws_send(c, "hello", 5, WEBSOCKET_OP_TEXT);
    }
    else if (ev == MG_EV_WS_MSG)
    {
        // When we get echo response, print it
        struct mg_ws_message *wm = (struct mg_ws_message *)ev_data;
        printf("WS message: [%.*s]\n", (int)wm->data.len, wm->data.ptr);
    }

    if (ev == MG_EV_ERROR || ev == MG_EV_CLOSE) // || ev == MG_EV_WS_MSG)
    {
        *(bool *)fn_data = true; // Signal that we're done
    }
}

int main(int argc, char *argv[])
{
    struct mg_mgr mgr;       // Event manager
    bool done = false;       // Event handler flips it to true
    struct mg_connection *c; // Client connection

    printf("polygon.io WebSocket client.\n");

    // handle termination signals
    signal(SIGINT, signal_handler);
    signal(SIGTERM, signal_handler);

    // an infinite loop, broken by a termination signal
loop:

    mg_mgr_init(&mgr); // Initialise event manager
    mg_log_set("4");
    c = mg_ws_connect(&mgr, s_url, fn, &done, NULL); // Create client

    while (c && done == false)
        mg_mgr_poll(&mgr, 1000); // Wait for echo

    mg_mgr_free(&mgr); // Deallocate resources

    if (s_signo == 0)
    {
        sleep(1);
        goto loop;
    }

    printf("\npolygon.io shutdown completed.\n");
}
```
- My expectation was: proceed to the "MG_EV_WS_OPEN" event
- The result I saw: connection errors
chris@pop-os:~/Phi/FX/Polygon$ ./polygon 
polygon.io WebSocket client.
162d008 3 mongoose.c:2761:mg_connect    1 -1 wss://socket.polygon.io/forex
162d008 3 mongoose.c:2761:mg_connect    2 -1 udp://8.8.8.8:53
162d008 3 mongoose.c:3494:mg_connect_re 2 0x3
162d008 4 mongoose.c:361:mg_sendnsreq   1 resolving socket.polygon.io @ 8.8.8.8, txnid 1
162d008 3 mongoose.c:3295:mg_send       2 0x3 0:0 35 err 0 (Success)
162d019 4 mongoose.c:3675:mg_mgr_poll   2 r- tchrc
162d019 3 mongoose.c:3405:read_conn     2 0x3 0:0 51 err 0 (Success)
162d019 4 mongoose.c:266:dns_cb         socket.polygon.io 1
162d019 3 mongoose.c:277:dns_cb         1 socket.polygon.io is 198.44.194.17
162d019 3 mongoose.c:3494:mg_connect_re 1 0x4
162d019 4 mongoose.c:3675:mg_mgr_poll   1 -- tChrc
162d0d5 4 mongoose.c:3675:mg_mgr_poll   2 -- tchrc
162d0d5 4 mongoose.c:3675:mg_mgr_poll   1 -w tChrc
162d0d6 3 mongoose.c:4484:mg_tls_init   1 Setting TLS, CA: ca.pem, cert: null, key: null
162d147 3 mongoose.c:4561:mg_tls_init   1 SSL client OK
162d202 4 mongoose.c:3675:mg_mgr_poll   2 -- tchrc
162d202 4 mongoose.c:3675:mg_mgr_poll   1 r- TcHrc
40F709B2C57F0000:error:0A000086:SSL routines:tls_post_process_server_certificate:certificate verify failed:../ssl/statem/statem_clnt.c:1883:
162d205 1 mongoose.c:404:mg_error       1 0x4 tls hs: rc -1, err 1
162d205 1 polygon.c:37:fn               0x4 tls hs: rc -1, err 1
162d205 3 mongoose.c:2738:mg_close_conn 1 closed
162d205 4 mongoose.c:3675:mg_mgr_poll   2 -- tchrC
162d205 3 mongoose.c:2738:mg_close_conn 2 closed
162d205 3 mongoose.c:2817:mg_mgr_free   All connections closed
162d5ee 3 mongoose.c:2019:mg_log_set    Setting log level to 4
162d5ee 3 mongoose.c:2761:mg_connect    1 -1 wss://socket.polygon.io/forex
162d5ee 3 mongoose.c:2761:mg_connect    2 -1 udp://8.8.8.8:53
162d5ee 3 mongoose.c:3494:mg_connect_re 2 0x3
162d5ee 4 mongoose.c:361:mg_sendnsreq   1 resolving socket.polygon.io @ 8.8.8.8, txnid 1
162d5ee 3 mongoose.c:3295:mg_send       2 0x3 0:0 35 err 0 (Success)
162d5ee 4 mongoose.c:3675:mg_mgr_poll   2 -- tchrC
162d5ee 1 mongoose.c:404:mg_error       1 0xffffffffffffffff DNS error
162d5ee 1 polygon.c:37:fn               0xffffffffffffffff DNS error
162d5ee 3 mongoose.c:2738:mg_close_conn 2 closed
162d5ee 4 mongoose.c:3675:mg_mgr_poll   1 -- tchRC
162d5ee 3 mongoose.c:2738:mg_close_conn 1 closed
162d5ee 3 mongoose.c:2817:mg_mgr_free   All connections closed
^C162d745 3 mongoose.c:2019:mg_log_set    Setting log level to 4
162d745 3 mongoose.c:2761:mg_connect    1 -1 wss://socket.polygon.io/forex
162d745 3 mongoose.c:2761:mg_connect    2 -1 udp://8.8.8.8:53
162d745 3 mongoose.c:3494:mg_connect_re 2 0x3
162d745 4 mongoose.c:361:mg_sendnsreq   1 resolving socket.polygon.io @ 8.8.8.8, txnid 1
162d745 3 mongoose.c:3295:mg_send       2 0x3 0:0 35 err 4 (Interrupted system call)
162d745 4 mongoose.c:3675:mg_mgr_poll   2 -- tchrC
162d745 1 mongoose.c:404:mg_error       1 0xffffffffffffffff DNS error
162d745 1 polygon.c:37:fn               0xffffffffffffffff DNS error
162d745 3 mongoose.c:2738:mg_close_conn 2 closed
162d745 4 mongoose.c:3675:mg_mgr_poll   1 -- tchRC
162d745 3 mongoose.c:2738:mg_close_conn 1 closed
162d745 3 mongoose.c:2817:mg_mgr_free   All connections closed

polygon.io shutdown completed.
- My question is: Am I doing anything wrong?

# Environment

- mongoose version: 7.6, the latest github master

jvo203, 2022-05-08:
As an extra information, this wss secure WebSocket connection works fine in Julia or golang, without any need to set ca.pem or any special SSL settings. Somehow mongoose seems to have problems with it, or perhaps I am using mongoose in an incorrect way.

jvo203, 2022-05-09:
Finally managed to solve it but the examples were a bit misleading. Instead of doing what the HTTPS client example suggests, the correct way with WSS seems to be this:

```
c = mg_ws_connect(&mgr, s_url, fn, &done, NULL); // Create client
struct mg_tls_opts opts = {.cert = "cert.pem", .certkey = "key.pem"};
mg_tls_init(c, &opts);

 while (c && (done == false) && (s_signo == 0))
    mg_mgr_poll(&mgr, 1000); // Wait for echo
``` 

The TLS stuff is enabled after creating the ```c``` connection, not in the ```MG_EV_CONNECT ```event, and both the certificate and key are passed (via ```.cert```, not ```.ca```).

At least this is what worked for me.

closed 2022-05-09 by jvo203

Mervyen, 2022-05-30:
> Finally managed to solve it but the examples were a bit misleading. Instead of doing what the HTTPS client example suggests, the correct way with WSS seems to be this:
>
> ```
> c = mg_ws_connect(&mgr, [...]

in the `mg_tls_opts`  I get the `.cert = ca.pem`(this file is finded in the /mongoose/examples/websocket-client/) 
how to get the `certkey`?  I get error in the `mg_tls_init` --> "Invalid SSL key, err 1"

thx for your replay

Mervyen, 2022-05-30:
thx，I should make my self too.