cesanta/mongoose #1548: Request with `Expect: 100-continue` header and chunked messages

smilediver, 2022-05-12:
There are situations, where web client sends only the request head with `Expect: 100-continue` header. In this scenario web server is supposed to reply with `100 Continue`, so that the client can proceed to send the body of the request. This request is handled by `MG_EV_HTTP_CHUNK` events. The problem is that the `mg_http_message` data in the first `MG_EV_HTTP_CHUNK` and the last `MG_EV_HTTP_CHUNK` are identical and there's no easy way to distinguish them.

I was wondering, would it be a bad practice, if after replying with `100 Continue`, I would modify the message in the recv buffer and remove `Expect: 100-continue` header?

Request example:
```
PUT /webdav/temp_5000B_file HTTP/1.1
Expect: 100-continue
Content-Length: 5120
Content-Type: application/octet-stream
Host: 127.0.0.1:3000
Connection: Keep-Alive
User-Agent: Cyberduck/8.3.3.37544 (Mac OS X/12.3.1) (x86_64)
Accept-Encoding: gzip,deflate
```

Processing example:
```
MG_EV_HTTP_CHUNK: 0x600003d000f0:29940 PUT /webdav/temp_5000B_file (body: 5120, chunk: 0)
<-- Need to reply here with 100
MG_EV_HTTP_CHUNK: 0x600003d000f0:29940 PUT /webdav/temp_5000B_file (body: 5120, chunk: 3766)
MG_EV_HTTP_CHUNK: 0x600003d000f0:29940 PUT /webdav/temp_5000B_file (body: 5120, chunk: 1354)
MG_EV_HTTP_CHUNK: 0x600003d000f0:29940 PUT /webdav/temp_5000B_file (body: 5120, chunk: 0)
```

cpq, 2022-05-14:
Not sure that it is generic enough to add the support for it into data structures...
Essentially we're looking for a simple flag - which can easily be done via `c->label`
Like,
```c
if (c->label[0] == '\0') c->label[0] = 'X'; // First chunk!
```

Or, `c->label` could store a total received counter:
```c
size_t *counter = (size_t *) c->label;
*counter += hm->chunk.llen;

if (*counter >= hm->message.len)  // Last chunk
```

Would that work?

smilediver, 2022-05-16:
Totally forgot about the label. Yeah, that would work, I'll probably end up inserting a structure there. Thanks!

Though, my biggest gripe, is that I would expect the library to pass this information to the user, especially when it already tracks and knows it. My guess is I'm not the only one reimplementing it.

What if `mg_http_message` would have something like `size_t remaining_len` that would indicate how much data is still expected to be received: `>0` indicates how much data is still expected to be received; `-1` indicates that it still expects data, but doesn't know how much; and `0` indicates that full request was received?

You can close this issue if you don't plan to add anything.

cpq, 2022-06-02:
Thank you.
Closing this without changing the API.

closed 2022-06-02 by cpq