cesanta/mongoose #1550: `mg_open_listener()` on windows fails with

smilediver, 2022-05-13:
`mg_open_listener()` on Windows 11 fails with `Error code 10022 is WSAEINVAL : Invalid argument` when trying to apply `SO_EXCLUSIVEADDRUSE` option.

I think this happens because both `SO_REUSEADDR` and `SO_EXCLUSIVEADDRUSE` are applied at the same time, and only one is supposed to be set at a time. Probably the define `(MG_ARCH == MG_ARCH_WIN32)` for `SO_REUSEADDR` is actually supposed to be `(MG_ARCH == MG_ARCH_WIN32 && !defined(SO_EXCLUSIVEADDRUSE))`.

richardsimaging, 2022-05-19:
Same problem here.

OgreTransporter, 2022-05-20:
Same problem here.

yalanmao, 2022-05-20:
(MG_ARCH == MG_ARCH_WIN32 && !defined(SO_EXCLUSIVEADDRUSE))  problem solved thanks @[smilediver](https://github.com/smilediver)

marcelwirtz, 2022-05-20:
Looks like i am facing the same problem on Windows 10 after upgrading to 7.7.

```
f6b27a 1 mongoose.c:4177:mg_open_listen exclusiveaddruse: 10022
f6b27a 1 mongoose.c:3571:mg_listen      Failed: 0.0.0.0:62839, errno 0
```

Where i have to change define?

smilediver, 2022-05-20:
https://github.com/cesanta/mongoose/blob/7.7/mongoose.c#L4156

cpq, 2022-05-21:
Can someone file a PR please.
I'll integrate and move 7.7 tag to include that fix into 7.7

closed 2022-05-21 by cpq

smilediver, 2022-05-23:
The fix was supposed to be where `setsockopt()` is called for `SO_REUSEADDR`, and not `SO_EXCLUSIVEADDRUSE`, ie it was supposed to end up here:

```
#if ((MG_ARCH == MG_ARCH_WIN32) || (MG_ARCH == MG_ARCH_UNIX) || \
     (defined(LWIP_SOCKET) && SO_REUSE == 1))
    } else if (setsockopt(fd, SOL_SOCKET, SO_REUSEADDR, (char *) &on,
```

cpq, 2022-05-23:
@smilediver thank you. Can you show a diff against the current HEAD please?

commit d5993ba2 (2022-05-21): Fix #1550 - excladdrinuse