cesanta/mongoose #1551: `mg_timer_add` calling function twice.

npodonnell, 2022-05-15:
Hi. 

I'm having an issue with `mg_timer_add`. This doesn't seem like correct behavior:

Code:
```c
#include "mongoose.h"

static const int poll_interval_ms = 50;

struct state {
    bool running = false;
    struct mg_mgr mgr;
};

void init(void* data) {
    struct state* ss = (struct state*) data;
    MG_INFO(("init called"));
} 

int main() {
    struct state ss;

    mg_mgr_init(&ss.mgr);

    // Schedule init to run once. This kicks everything off.
    mg_timer_add(&ss.mgr, 0, MG_TIMER_RUN_NOW, init, &ss);

    while(true) {
        mg_mgr_poll(&ss.mgr, poll_interval_ms);
    }

    mg_mgr_free(&ss.mgr);
}
```

Output:
```
10f6756 2 main.c:12:init       init called
10f6756 2 main.c:12:init       init called
```

I want to call `init` once at the very beginning, but it's called twice. If I set the `ms` parameter to something > 0, (e.g. 5000) it will call init immediately, then after 5s.

cpq, 2022-05-15:
Please fetch HEAD and try again

cpq, 2022-05-19:
Fixed - closing this. Please reopen if you think it is still an issue.

closed 2022-05-19 by cpq

npodonnell, 2022-05-21:
ty