cesanta/mongoose #1559: a non-blocking "send()" in the multi-threaded example

jvo203, 2022-05-22:
- My question is: The new multi-threaded example sends "hi" using ```send()```: ```send(sock, "hi", 2, 2);           // Wakeup event manager```. The question is, what is the meaning of the last parameter flag "2" passed to ```send()```? Ideally I would like to send the message in a non-blocking way. The prior mongoose 7.6 defined MSG_NONBLOCKING as 0. 7.7 also uses 0 internally in some places. So what would "1" and "2" mean?

# Environment

- mongoose version: 7.7

cpq, 2022-05-22:
@jvo203 that was a typo, fixed. The last parameter should be 0 (no flags, block).  To be entirely correct, a code should also check for a return code and send the remaining data if not everything was sent. In practice, small sends are smaller than a socket buffer so send() sends everything.

A send() should be blocking. There is no reason to have a non-blocking send in a worker thread: here, you're sending results of your work to the connected peer. Can't loose your data.

closed 2022-05-22 by cpq

jvo203, 2022-05-22:
I agree in normal workloads the send() should be blocking. However, in my particular case I can afford to lose some data. I am streaming a large amount of time series data (time snapshots), and it's OK to lose some of them since in a few seconds a newer time snapshot will be available anyway.

So I guess using "MSG_DONTWAIT" instead of "0" would implement a non-blocking but potentially lossy "send()".

cpq, 2022-05-23:
Yes. You could also manually put a socket in a non-blocking mode.