cesanta/mongoose #1560: Feature Request: Content Negotiation

EndlessDex, 2022-05-23:
Would love to see support for [content negotiation](https://developer.mozilla.org/en-US/docs/Web/HTTP/Content_negotiation) added.

Specifically the ability to use the [Accept-Encoding](https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Accept-Encoding)/[Content-Encoding](https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Content-Encoding) headers to negotiate compression schemes.

Work would be three parts mainly:
- add configuration options to:
  - Enable encoding
  - Specify which files extensions to encode (ex. encode only html, css, and js  r"\.(js|css|html)$" )
  - Specify which types of encoding are available (gzip,br,etc)
- add parsing for negotiation messages (read browser supported encoding and write selected encoding to header)
- modify target filepath when encoding scheme is in use to point to encoded files (add appropriate extension index.html.br)

Some webservers support dynamic encoding of files at serve-time, but I think that would be beyond the scope of an embedded server like mongoose. Static serving of compressed files would be a really nice addition to save space on the server machine as well as improve load times of pages.

[image]

cpq, 2022-05-23:
@EndlessDex thank you.

Could you elaborate on a practical use case for it please?

EndlessDex, 2022-05-23:
Almost all major websites use some form of file compression for their html/script files. Below is an example from github itself. Not exactly sure what you mean by "practrical use case"? Compression for file transmission is always practical!

[image]

cpq, 2022-05-24:
Agree that saves on latency and also on a precious space on embedded systems.

What we could do is this: when a request to a static file X comes, and
- `Accept-Encoding: gzip` is specified
- file `X` is not present
- file `X.gz` is present
Then mongoose simply serves the `.gz` file with the extra `Content-Encoding: gzip` header

Parsing all sorts of different encodings and adding complex configuration sounds like an overkill - unless there are strong counter-arguments.

cpq, 2022-06-02:
@EndlessDex that's done, please test. Note that this change ignores the value of `Accept-Encoding` header.
Normally, we should honor it and if the `gzip` is not set, serve a non-gzipped content. Mongoose cannot do that, since it cannot gunzip on the fly. So we serve anyways, hoping that client can gunzip.

commit fe682f83 (2022-06-02): Fix #1560 - implement precompressed .gz file serving
closed 2022-06-02 by cpq