cesanta/mongoose #1562: in the new multi-threaded framework, the pipe event manager does not get woken up for each message

jvo203, 2022-05-25:
- My goal is: To produce a JSON message in a "producer" thread, send the message via a Unix socket to the pipe event manager, have the pipe callback called once per each message
- My actions were: The producer thread sends "{json element}" ```send(channel, json, strlen(json), MSG_DONTWAIT); // Wakeup event manager, flags: 0 - blocking, MSG_DONTWAIT - non-blocking```. The "consumer" thread (a pipe callback) usually receives the sent JSON "{json element}". But sometimes it receives two successive JSON objects in one concatenated string "{json element 1}{json element 2}". This happens when the producer "send()" calls happen in a very quick succession.
- My expectation was: After sending the JSON using send() in the producer thread, the pipe event manager would wake up and call the pipe callback function (exactly *ONCE* per *EACH* message)
- The result I saw: Sometimes the Unix socket messages are collated and the pipe callback is called once for the combined messages (more than one JSON message).
- My question is: Is this a bug/undocumented feature or simply an unintended consequence of using Unix pipes in mongoose v7.7 versus UDP in the prior v7.6?

- My workaround: I have resorted to appending a pipe character '|' to each JSON message and then splitting the received WebSocket payload on the client in JavaScript as per below:

"producer" thread (it does not matter if the send flag is set to 0 or MSG_DONTWAIT, either way the problem persists)
```
// prepare a JSON response (notice a pipe separator at the end of the string)
char *json = mjson_aprintf("{%Q:%Q,%Q:%ld,%Q:%g,%Q:%g,%Q:%g,%Q:%g}|", "pair", pair, "lTid", lTid, "bid", bid, "ask", ask, "signal", 0.0, "cumulative", 0.0);

// pass the JSON string over to the mongoose web server
send(channel, json, strlen(json), MSG_DONTWAIT); // Wakeup event manager, flags: 0 - blocking, MSG_DONTWAIT - non-blocking

free(json);
```

"consumer" thread (pipe callback)
```

// Pipe event handler
static void pipe_cb(struct mg_connection *c, int ev, void *ev_data, void *fn_data)
{
    // broadcast the data to all WebSocket connections
    if (ev == MG_EV_READ)
    {
        struct mg_connection *t;
        for (t = c->mgr->conns; t != NULL; t = t->next)
        {
            // do not bother comparing strings for non-WebSocket connections
            if (t->is_websocket)
            {
                // printf("[C] found a WebSocket connection, sending %zu bytes.\n", msg->len);
                mg_ws_send(t, c->recv.buf, c->recv.len, WEBSOCKET_OP_TEXT);
            }
        }

        c->recv.len = 0; // Tell Mongoose we've consumed data
    };
}
```
client-side web browser JavaScript
```
conn.addEventListener("message", function (evt) {
        var msg = evt.data;

        // there may be more than one JSON message
        var elements = msg.split('|');

        elements.forEach(function (val) {
            try {
                payload = JSON.parse(val);
                process_payload(payload);
            } catch (_) { }
        })
    });
```
The workaround works fine but it does not deal with the root cause of the problem.

# Environment

- mongoose version: 7.7

jvo203, 2022-05-25:
In other words, I expected to see the following timeline of events:
```
1. send(channel, json, strlen(json), MSG_DONTWAIT); 
2. pipe_cb(struct mg_connection *c, int ev, void *ev_data, void *fn_data) is called with "{json msg 1}"
3. send(channel, json, strlen(json), MSG_DONTWAIT); 
4. pipe_cb(struct mg_connection *c, int ev, void *ev_data, void *fn_data) is called with "{json msg 2}"
5. 1. send(channel, json, strlen(json), MSG_DONTWAIT); 
6. pipe_cb(struct mg_connection *c, int ev, void *ev_data, void *fn_data) is called with "{json msg 3}"
```
But instead the following timeline is observed:

```
1. send(channel, json, strlen(json), MSG_DONTWAIT); 
2. pipe_cb(struct mg_connection *c, int ev, void *ev_data, void *fn_data) is called with "{json msg 1}"
3. send(channel, json, strlen(json), MSG_DONTWAIT); 
4. pipe_cb(struct mg_connection *c, int ev, void *ev_data, void *fn_data) is called with "{json msg 2}"
5. 1. send(channel, json, strlen(json), MSG_DONTWAIT); 
6. pipe_cb is not called
7. 1. send(channel, json, strlen(json), MSG_DONTWAIT); 
8. pipe_cb(struct mg_connection *c, int ev, void *ev_data, void *fn_data) is called with "{json msg 3}{json msg 4}"
9. 1. send(channel, json, strlen(json), MSG_DONTWAIT); 
10. pipe_cb(struct mg_connection *c, int ev, void *ev_data, void *fn_data) is called with "{json msg 5}"
```

Messages are not lost, just some of them are concatenated and handled in a single "pipe_cb" event call. I would have expected each and every message to be handled in a separate "pipe_cb" event call, with no strange event concatenation.

cpq, 2022-05-30:
@jvo203 sending one wake-up event per call is possible if a connected socketpair used UDP.
Currently it uses TCP.

However, using UDP has a danger of loosing messages is producer is too quick. That is not acceptable.

I suggest to separate JSON messages by some sort of delimiter - for example, a new line.
A receiver side would check c->recv.buf, and if a delimiter exists, it'd send a message and then delete it:

```c
uint8_t *p = (uint8_t *) memchr(c->recv.buf, '\n', c->recv.len);
if (p != NULL) {
   size_t n = p - c->recv.buf;
   send_message_to_the_client(client_connection, c->recv.buf, n);
   mg_iobuf_del(&c->recv, 0, n);
}
```

Would that work?

jvo203, 2022-05-31:
Yes, using a delimiter along the lines of your suggestion with '\n' would work fine with JSON messages. As explained in my message, I currently use a pipe delimiter '|' but without any server-side message splitting (this is done in JavaScript client-side).

However, my other application uses binary messages and this is where things would get a bit "messy" with any sort of delimiter characters. For example, one would need to send the binary message length (4 or 8 bytes, likely size_t) prior to the main binary payload (this way the mongoose receiver side knows the size of each binary "sub-message"). Or, as I currently do in another application (which still uses mongoose v7.6 with a UDP socketpair), send binary payloads as structures (sizeof(struct XXX)). The size of each binary structure is known in advance, and the number of "sub-messages" would be obtained from ```c->recv.len / sizeof(struct XXX)```.

Anyway, ideally perhaps it would be possible to switch between UDP and TCP socket pairs by passing a user parameter when creating the mg socket pair. Right now I am reluctant to upgrade the main application from mongoose v7.6 to v7.7 precisely because suddenly there might be more than one binary message per one wake-up event call, which the current code would struggle to handle. What is fine with the text-based JSON might not be acceptable in the "binary" world.

cpq, 2022-05-31:
Noted, thank you @jvo203 . Added an extra param to mg_mkpipe. Please pull and try.

jvo203, 2022-06-01:
Thank you, have tested the new ```mg_mkpipe()``` in the text JSON application. It works as "expected": the UDP mode gives one message per one wake-up call, and the TCP mode sometimes produces two concatenated messages in a single wake-up call. Will try to migrate the other (binary messaging) application from mongoose v7.6 to v7.7 too. Am closing this issue for now (if any problems re-surface the issue can always be re-opened). Thank you.

By the way, I have noticed mongoose.c contains an internal function ```static void mg_set_non_blocking_mode(SOCKET fd)``` that puts a socket into a non-blocking mode. However, this function is not exposed in the ```mongoose.h``` header. So for now I have been using this piece of code:
```
int guard(int n, char *err)
{
    if (n == -1)
    {
        perror(err);
        exit(1);
    }
    return n;
}

// put it into a non-blocking mode
int flags = guard(fcntl(channel, F_GETFL), "could not get file flags");
guard(fcntl(channel, F_SETFL, flags | O_NONBLOCK), "could not set file flags");
``` 

Just a thought, perhaps it might make sense to make ```static void mg_set_non_blocking_mode(SOCKET fd)``` visible to end-users too.

jvo203, 2022-06-01:
Is this needed too? ```fcntl(fd, F_SETFD, FD_CLOEXEC);                          // Set close-on-exec```
You have it in your ```mg_set_non_blocking_mode()``` function.

jvo203, 2022-06-01:
Unfortunately, after updating the binary messaging application from mongoose 7.6 to 7.7, the binary messages are received in a concatenated form even for a UDP connection. This happens quite often.

a sender thread:
```
// create a UDP message
struct websocket_message msg = {strdup(resp->session_id), payload, msg_len};

// pass the message over to mongoose via a communications channel
ssize_t sent = send(channel, &msg, sizeof(struct websocket_message), 0); // Wakeup event manager
```

a receiver end:
```[C] mg_pipe_callback::abort (size mismatch): expected: 24, received: 696!```

```696 / 24``` = 29 --> 29 messages have been concatenated within one binary buffer. I can get around it by trying to handle all those 29 messages sequentially in a for loop (the binary size of each message is always the same, equal to 24 bytes).

So I guess the issue of messages being concatenated is not really related to differences between UDP and TCP. It is probably caused by re-designed event polling in mongoose 7.7 versus 7.6. In mongoose v7.6 there was exactly one message per one wake-up event. In v7.7 this one-to-one correspondence is gone even for UDP sockets.

jvo203, 2022-06-01:
Am closing this issue, this time it was my mistake. When migrating the binary messaging application to mongoose v7.7, I forgot to add ```c->recv.len = 0; // Tell Mongoose we've consumed data```, which resulted in the messages quickly accumulating in mongoose buffers. This code was not needed by v7.6.

closed 2022-06-01 by jvo203

jvo203, 2022-06-01:
OK, I'll stick to my existing code for putting the socket into a non-blocking mode. The binary messages application has been migrated successfully to mongoose v7.7, and it now works with both UDP and TCP mongoose sockets. It has been made to accept / process multiple concatenated messages like this:
```
n = c->recv.len / sizeof(struct websocket_message);
printf("[C] mg_pipe_callback: received %d binary message(s).\n", n);

size_t offset = 0;
for (i = 0; i < n; i++)
{
  struct websocket_message *msg = (struct websocket_message *)(c->recv.buf + offset);
  offset += sizeof(struct websocket_message);

...
```
No problems, the code works fine with either 1 or more than 1 messages (luckily each binary message is of the same size so it is easy to calculate the number messages and message boundaries).