cesanta/mongoose #1582: Support disabling logs from binary

david-collett, 2022-06-08:
- My goal is: A facility for completely omitting log strings from a binary build with mongoose.
- My actions were: define MG_ENABLE_LOG=0
- My expectation was: log messages would be excluded
- The result I saw: log messages are present.
- My question is:

Since the recent removal of MG_ENABLE_LOG, it seems no longer possible to produce a binary without logging strings embedded in it. Even if you define PUTCHAR(x) to nothing, the log strings will still be present, increasing the size of the binary and serving no purpose.

It would be great if MG_ENABLE_LOG was reinstated, or some other mechanism was provided to disable logging at the preprocessor stage so that strings do not end up in the binary.

# Environment

- mongoose version: master
- Compiler/IDE and SDK: ZZZ version _X.Y_
- Target hardware/board: _?_
- Connectivity chip/module: _?_

cpq, 2022-06-08:
@david-collett thank you
Leaking logs is definitely an issue - could you provide more data, what exactly gets leaked? Some example output?

david-collett, 2022-06-08:
Hi, consider the following example involving a small change to an example to represent a "production" build:

```
dave@morty:~/src/mongoose/examples/http-client$ git diff
diff --git a/examples/http-client/main.c b/examples/http-client/main.c
index 2ff3ca8a..5e33e606 100644
--- a/examples/http-client/main.c
+++ b/examples/http-client/main.c
@@ -9,6 +9,7 @@
 //    make MBEDTLS=/path/to/your/mbedtls/installation
 //    make OPENSSL=/path/to/your/openssl/installation
 
+#define MG_PUTCHAR(x)
 #include "mongoose.h"
 
 // The very first web page in history. You can replace it from command line
@@ -62,7 +63,7 @@ int main(int argc, char *argv[]) {
   struct mg_mgr mgr;              // Event manager
   bool done = false;              // Event handler flips it to true
   if (argc > 1) s_url = argv[1];  // Use URL provided in the command line
-  mg_log_set("4");                // Set to 0 to disable debug
+  mg_log_set("0");                // Set to 0 to disable debug
   mg_mgr_init(&mgr);              // Initialise event manager
   mg_http_connect(&mgr, s_url, fn, &done);  // Create client connection
   while (!done) mg_mgr_poll(&mgr, 50);      // Infinite event loop
dave@morty:~/src/mongoose/examples/http-client$ make example
cc ../../mongoose.c -I../.. -W -Wall -DMG_ENABLE_LINES  -o example main.c
dave@morty:~/src/mongoose/examples/http-client$ strip example
dave@morty:~/src/mongoose/examples/http-client$ strings example | grep resolving
%lu resolving %.*s @ %s, txnid %hu
dave@morty:~/src/mongoose/examples/http-client$ 
```

That string is from an `MG_VERBOSE` call in `mg_sendnsreq`. It is still present despite `mg_log_set("0")` and setting `#define MG_PUTCHAR(x)` prior to including mongoose.h. Too much is determined at runtime for the compiler to optimise the strings away.

I think the solution requires a compile-time define to make MG_LOG get defined away (to nothing). I can make a pull-request to this effect if you like.

Dave

cpq, 2022-06-10:
Thank you @david-collett 

The recent change to the http-client example allows to set the log level via the environment variable `LOG_LEVEL`.
Here's my test with the level 4:

```sh
$ make -s -C examples/http-client/  LOG_LEVEL=4
1814d65721e 3 net.c:183:mg_connect      1 -1 http://info.cern.ch/
1814d65721e 3 net.c:183:mg_connect      2 -1 udp://8.8.8.8:53
1814d65721e 4 dns.c:250:mg_sendnsreq    1 resolving info.cern.ch @ 8.8.8.8, txnid 1
...
```
And with level 0:

```sh
$ make -s -C examples/http-client/  LOG_LEVEL=0
HTTP/1.1 200 OK
Date: Fri, 10 Jun 2022 11:35:33 GMT
...
```

Meaning, I cannot reproduce your issue. With log level 0, no logs are printed.
Can you clarify you're seeing the same, please?

david-collett, 2022-06-14:
The strings are not printed, but they are still present in the binary (as shown by the "strings" command in my example). Best practice IMO is that when debug output is disabled, the strings (and the accompanying debug code) is not built at all, and hence doesn't appear in the binary, however you may disagree. I will make a small patch to show you what I mean.

david-collett, 2022-06-14:
Honestly I'd rather just see bc16854ccb359a0d2323a2befd4739fc720f737e and 7e2ca72310d81ac05324319c8e3bf33dc5c552f1 reverted. What was the reason for these?

cpq, 2022-06-19:
I agree that MG_ENABLE_LOG=0 should be restored.
W.r.t. redirecting logs, how about making MG_PUTCHAR a re-settable callback?
Yes, it will print a single character a time, but it is easy to buffer it, e.g.:

```c
static void logging_putchar(uint8_t ch) {
  static char buf[128];
  static struct mg_iobuf log = { .buf = buf, .size = sizeof(buf), .len = 0};
  log.buf[log.len++] = ch;
  if (ch == '\n' || log.len >= log.size) {
    printf("%.*s", (int) log.len, log.buf);  // Flush buffer. Could be syslog
    log.len = 0;
  }
}
```

Why this? Because in the mongoose log function, a single log like is broken
down to three separate outputs: prefix (file name, line num, time), actual log string, and a newline.
With `mg_hexdump()` there are even more separate outputs. That's why the putchar interface is chosen. 

Would the approach above work for you?

cpq, 2022-06-19:
Pushed https://github.com/cesanta/mongoose/commit/9760b7c9d6ce7bb351502660f7779e6340e554d1 - please feedback

david-collett, 2022-06-20:
Apologies for the delay, this looks good, thanks.

closed 2022-06-20 by cpq