cesanta/mongoose #1590: websocket receive string buff not cleared .

matthiashack, 2022-06-14:
for example:  first time the client send string length is 1000 , second time the websocket client send length is 10, I got the error content sometimes.   I guess maybe  the buff not cleared sometimes

matthiashack, 2022-06-14:
I tried this `mg_iobuf_del(&c->recv, 0, 1000);` but not work.

matthiashack, 2022-06-15:
[image]

scaprile, 2022-06-15:
Messages have a length indication. May I ask what are you doing ? Please provide example code where to see your issue.

matthiashack, 2022-06-16:
@scaprile 
I am using the  mongoose as a websocket server, communicate with the Unity game client and the webrtc browser page.  here is the code I am receiving message: 
`if (ev == MG_EV_WS_MSG) {
		//receive message
		struct mg_ws_message* wm = (struct mg_ws_message*)ev_data;
		std::string cli_msg = wm->data.ptr;
              //do some process    
}`

the client send the  data very fast, about 100 records per second, I will receive the wrong data occasionally.

matthiashack, 2022-06-17:
hi @scaprile,  thank you very much ,you resolve my problem.

closed 2022-06-17 by matthiashack