cesanta/mongoose #1628: mg_sntp_parse misinterprets fractional seconds, causing timestamp errors averaging 1/3 second

jbrzusto, 2022-07-13:
quoting from [RFC-5905](https://datatracker.ietf.org/doc/html/rfc5905#page-12):

>    The 64-bit timestamp format is used in packet headers and other
>    places with limited word size.  It includes a 32-bit unsigned seconds
>    field spanning 136 years and a 32-bit fraction field resolving 232
>    picoseconds.  

But `mg_sntp_parse` treats the fractional field as if it is a `uint32_t` count number of microseconds: 
```
unsigned long useconds = mg_ntohl(data[1])
res = ((int64_t) seconds) * 1000 + (int64_t) ((useconds / 1000) % 1000);
```
In fact, `(useconds / 1000) % 1000` represents a portion of the fractional seconds which has a sub-millisecond total range, and is effectively random with respect to the true milliseconds.  So the fractional seconds returned by `mg_sntp_parse` are off by an average of 1/3 second (the mean absolute difference between two independent, uniformly distributed variables over the range [0, 1))

cpq, 2022-08-05:
@jbrzusto I guess this can be closed now, as your commit fixes it?

cpq, 2022-08-09:
Closing this, please reopen if required.

closed 2022-08-09 by cpq