cesanta/mongoose #1635: How do I close a TCP connection?

tilongzs, 2022-07-19:
- My goal is: The server gets the message that the client is disconnected from the server.
- My actions were: mg_close_conn(_currentConn);
- My expectation was: The client executes mg_close_conn(conn); mg_mgr_free(mgr); code, and the server receives the MG_EV_CLOSE event.
- The result I saw: The client executes mg_close_conn(conn); mg_mgr_free(mgr); code, the client receives the MG_EV_CLOSE event, but the server does not receive the MG_EV_CLOSE event. The server side receives the MG_EV_CLOSE event only when the client program is completely closed.
- My question is: Which step am I missing?

# Environment

- mongoose version: 7.7
- Compiler/IDE and SDK: Visual Studio 2022 (v143)  Windows SDK 10.0.19041.0
- Target hardware/board: Windows 11 X64(10.0.22000.795)

scaprile, 2022-07-19:
Please correct me where my guessing was off.
You have a server and a client on a TCP port. Both implemented using Mongoose, and hopefully running on different devices <sup>1</sup>.
Mongoose function calls mostly "request" an action to be done, the action itself gets actually done by the event manager, particularly on TCP where there are handshakes to be performed. So, you can't stop calling the manager and free its resources until you are sure your job is done.
<sup>1</sup> (If both client and server are on the same device and your client exits, your server might not have processed the event when you exit the event manager)

A proper client disconnect can be observed on the [HTTP client example](https://github.com/cesanta/mongoose/tree/master/examples/http-client) and is explained in detail in the [error handling tutorial](https://mongoose.ws/tutorials/error-handling/https://mongoose.ws/tutorials/error-handling/)

tilongzs, 2022-07-20:
> Please correct me where my guessing was off. You have a server and a client on a TCP port. Both implemented using Mongoose, and hopefully running on different devices 1. Mongoose function calls mostly "request" an action to be done, the action itself gets actually done by the event manager, particularly on TCP where there are handshakes to be performed. So, you can't stop calling the manager and free its resources until you are sure your job is done. 1 (If both client and server are on the same device and your client exits, your server might not have processed the event when you exit the event manager)
> 
> A proper client disconnect can be observed on the [HTTP client example](https://github.com/cesanta/mongoose/tree/master/examples/http-client) and is explained in detail in the [error handling tutorial](https://mongoose.ws/tutorials/error-handling/https://mongoose.ws/tutorials/error-handling/)

I re-ran the test, both sides are using Mongoose, when the server and client are running on different devices, the server can receive the MG_EV_CLOSE event. When running on the same device, the server cannot receive the MG_EV_CLOSE event.
When running on the same device: the server side uses Mongoose and the client side does not use Mongoose, the server side can receive MG_EV_CLOSE event; when the server side does not use Mongoose and the client side uses Mongoose, the server side cannot receive the connection disconnect message.
My requirement is simple, how can the client completely disconnect TCP connection without close the client program on the same device (as said before, I have executed mg_close_conn(conn); mg_mgr_free(mgr)).

tilongzs, 2022-07-21:
Give up.

tilongzs, 2022-07-21:
Give up.

closed 2022-07-21 by tilongzs