cesanta/mongoose #1648: mg_http_serve_file() path length limit is 80 chars ?

linuxguy123, 2022-07-26:
// add the server directory to the filename
       char fullFileURI[256];
        strcat(fullFileURI, SERVERDIR);
        strcat(fullFileURI,fileURI); 
        printf("Full file URI is: %s \n", fullFileURI); 

        // send the requested file
        mg_http_serve_file(c, hm, fullFileURI, &opts); 

This fails for a long file path, for example: "/home/me/Development/MissionControl/missionClient/three.js-master/build/three.module.js"  

The file requested is located as per the path.  The file permissions are fine.   Is there a length limit on the path ?   Seems to be about 80 chars ?

How does one put a code block into this editor ?

scaprile, 2022-07-26:
Works OK, please test your underlying filesystem or provide more information
[test.zip](https://github.com/cesanta/mongoose/files/9193852/test.zip)

linuxguy123, 2022-07-26:
So you created a file named three.module.js and put it in /home/me/Development/MissionControl/missionClient/three.js-master/build and it served it ?

scaprile, 2022-07-27:
The code attached in the former comment compiled and served the included file at the path you posted, under Linux, using defaults (POSIX FS).
What I seem to understand from your transaction details is that the client does receive a reply (200 OK) and some part of the file seems to be transferred (56K ?). I would check with Wireshark to see if everything is being sent; raise the debug level, might show you something; try running the test.zip I posted as is, then use your file instead.

linuxguy123, 2022-07-27:
I'll dig deeper.

cpq, 2022-08-04:
The path limit is driven by the `MG_PATH_MAX` constant, which can set by the arch-dependent header. If it is not set, then it defaults to 128:

```
$ grep  MG_PA src/arch*
src/arch_esp32.h:#define MG_PATH_MAX 128
src/arch_esp8266.h:#define MG_PATH_MAX 128
src/arch_freertos_lwip.h:#ifndef MG_PATH_MAX
src/arch_freertos_lwip.h:#define MG_PATH_MAX 128
src/arch_newlib.h:#define MG_PATH_MAX 100
src/arch_rtx_lwip.h:#ifndef MG_PATH_MAX
src/arch_rtx_lwip.h:#define MG_PATH_MAX 128
src/arch_win32.h:#define realpath(a, b) _fullpath((b), (a), MG_PATH_MAX)
```

cpq, 2022-08-04:
Actually on windows/unix we probably should use `FILENAME_MAX`. Or `PATH_MAX` ... I guess an ISO C constant is the way to go.

On the embedded, I remember cases when PATH_MAX was defined to some crazy high value, like 4k, so I guess it is safe to keep it low by default. It is always possible to override it.

@linuxguy123, try to add a build flag `-DMG_PATH_MAX=FILENAME_MAX` and see if that works for you.

commit e8289ec1 (2022-08-06): Fix #1648 - set MG_PATH_MAX to FILENAME_MAX on windows and unix
closed 2022-08-06 by cpq