cesanta/mongoose #1667: sorry unimplemented: non-trivial designated initializers not supported

afridali123, 2022-08-10:
getting this error while compiling code of http_client/ main.c

scaprile, 2022-08-10:
What is your compiler ? What version ? On what machine ? Under which OS ? For what target platform ?
Which file ? What line number ?
Can you please paste your command and the relevant error ?
Did you google for that error ? You seem to be trying to compile C (Mongoose and the example are C code) with a very old C++ compiler. If you have _clang_ try `make clean all CC=clang` (or the equivalent in your environment)

cpq, 2022-08-10:
@afridali123 an issue template is there for a reason. By neglecting it, you're causing extra time spent by everybody. Please don't.

afridali123, 2022-08-11:
its solved by updating compiler

closed 2022-08-11 by afridali123