cesanta/mongoose #1679: webui-login example does not support more than one http connections

liushuangit, 2022-08-19:
- My goal is: 
 If open multiple http login pages, the login page shows normal.
- My actions were: 
 when open the firt http web page, the login page shows normal,
 when I open the second http connection, the login page does not show, right click the mouse can see the index.html source code.
- My expectation was: 
each http connection can enter the login page.
- The result I saw: 
  when I open multiple login pages, the login page can not show.
- My question is: 
- how to let multiples login pages show normal as https connection? 

# Environment
- mongoose version: 7.7
- Compiler/IDE and SDK:  Linux gcc

scaprile, 2022-08-19:
In what scenario are you opening "multiple login pages" ? Your browser will cache the credentials as with any other web service. To try several simultaneous logins use different machines, different browsers, containers in Firefox, private windows in Firefox, incognito tabs in Chrome, etc; some way to isolate your environments so the browser does not already know your credentials.

scaprile, 2022-08-19:
Works for me

closed 2022-08-19 by scaprile