cesanta/mongoose #1736: Null pointer dereferences  (NULL_RETURNS) in mg_rpc_add()

jameshilliard, 2022-09-18:
I'm seeing this get flagged by coverity:
https://github.com/cesanta/mongoose/blob/0a265e79a67d7bfcdca27f2ccb98ccb474677ec6/src/rpc.c#L5
```c
Dereferencing "rpc", which is known to be "NULL".
```

scaprile, 2022-09-20:
Dereferencing occurs only at compile time to get the size of the struct

jameshilliard, 2022-09-23:
> Dereferencing occurs only at compile time to get the size of the struct

So should maybe change it to something like this then?:
```c
struct mg_rpc *rpc = (struct mg_rpc *) calloc(1, sizeof(struct mg_rpc)); 
```

jameshilliard, 2022-09-24:
For reference the analysis of coverity looks like this:
```c
3540void mg_rpc_add(struct mg_rpc **head, struct mg_str method,
3541                void (*fn)(struct mg_rpc_req *), void *fn_data) {
```
> 1. **returned_null: calloc returns NULL (checked 53 out of 61 times).**
> 2. **var_assigned: Assigning: rpc = NULL return value from calloc.**
```c
3542  struct mg_rpc *rpc = (struct mg_rpc *) calloc(1, sizeof(*rpc));
```
> CID 357183 (#1 of 1): **Dereference null return value (NULL_RETURNS)**
> 3. **dereference: Dereferencing rpc, which is known to be NULL.**
```c
3543  rpc->method = mg_strdup(method), rpc->fn = fn, rpc->fn_data = fn_data;
3544  rpc->next = *head, *head = rpc;
3545}
```

cpq, 2022-09-25:
sizeof(*TYPE) resolves to a correct value regardless of the allocation result .
If that's what a static analyzer complains about , then it is a static analyzer should be fixed instead.

closed 2022-09-25 by cpq

jameshilliard, 2022-09-25:
> sizeof(*TYPE) resolves to a correct value regardless of the allocation result .
> If that's what a static analyzer complains about , then it is a static analyzer should be fixed instead.

Hmm, maybe it's actually not complaining about `sizeof` but rather that the calloc return isn't checked against `NULL` in the event of an allocation failure?

PR #1765 (2022-09-25): Fix null pointer dereference in mg_rpc_add()
Fixes #1736
commit 3c71977e (2022-09-25): Fix null pointer dereference in mg_rpc_add()

Fixes #1736