cesanta/mongoose #1739: (TAINTED_SCALAR) in mg_ws_cb()

jameshilliard, 2022-09-18:
I'm seeing this get flagged by coverity:
https://github.com/cesanta/mongoose/blob/0a265e79a67d7bfcdca27f2ccb98ccb474677ec6/src/ws.c#L183
```c
Passing tainted expression "msg.data_len" to "mg_ws_send", which uses it as an offset.
```
https://github.com/cesanta/mongoose/blob/0a265e79a67d7bfcdca27f2ccb98ccb474677ec6/src/ws.c#L183
```c
Passing tainted expression "msg.data_len" to "mg_ws_send", which uses it as an offset.
```

scaprile, 2022-09-20:
The called function, `ws_process()`, has appropriate sanity checks in place

closed 2022-09-20 by scaprile

jameshilliard, 2022-09-23:
> The called function, `ws_process()`, has appropriate sanity checks in place

That's being flagged looks like, see analysis:
```c
5700static size_t ws_process(uint8_t *buf, size_t len, struct ws_msg *msg) {
5701  size_t i, n = 0, mask_len = 0;
5702  memset(msg, 0, sizeof(*msg));
```
> 1. Condition len >= 2, taking true branch.
```c
5703  if (len >= 2) {
5704    n = buf[1] & 0x7f;                // Frame length
```
> 2. Condition buf[1] & 0x80, taking true branch.
```c
5705    mask_len = buf[1] & 128 ? 4 : 0;  // last bit is a mask bit
5706    msg->flags = buf[0];
```
> 3. Condition n < 126, taking false branch.
```c
5707    if (n < 126 && len >= mask_len) {
5708      msg->data_len = n;
5709      msg->header_len = 2 + mask_len;
```
> 4. Condition n == 126, taking true branch.
> 5. Condition len >= 4 + mask_len, taking true branch.
```c
5710    } else if (n == 126 && len >= 4 + mask_len) {
5711      msg->header_len = 4 + mask_len;
```
> 6. **byte_swapping: Performing a byte swapping operation on buf[2] implies that it came from an external source, and is therefore tainted.**
> 7. **parm_assign: Assigning: msg->data_len = ((size_t)buf[2] << 8) | buf[3], which taints msg->data_len.**
```c
5712      msg->data_len = (((size_t) buf[2]) << 8) | buf[3];
```
> 8. Falling through to end of if statement.
```c
5713    } else if (len >= 10 + mask_len) {
5714      msg->header_len = 10 + mask_len;
5715      msg->data_len =
5716          (size_t) (((uint64_t) be32(buf + 2) << 32) + be32(buf + 6));
5717    }
5718  }
5719  // Sanity check, and integer overflow protection for the boundary check below
5720  // data_len should not be larger than 1 Gb
```
> 9. Condition msg->data_len > 1073741824UL /* 1024 * 1024 * 1024 */, taking false branch.
5721  if (msg->data_len > 1024 * 1024 * 1024) return 0;
> 10. Condition msg->header_len + msg->data_len > len, taking true branch.
```c
5722  if (msg->header_len + msg->data_len > len) return 0;
5723  if (mask_len > 0) {
5724    uint8_t *p = buf + msg->header_len, *m = p - mask_len;
5725    for (i = 0; i < msg->data_len; i++) p[i] ^= m[i & 3];
5726  }
5727  return msg->header_len + msg->data_len;
5728}
```