cesanta/mongoose #1749: DNS resolution exception

Jabriel828, 2022-09-21:
- My goal is: upload files
- My actions were: http post
- My expectation was: sucess
- The result I saw: Unexpected DNS response
- My question is: 
- domain :  upload.qiniup.com 
- Unable to parse normally

# Environment

- mongoose version: 7.8
- Target hardware/board: macbook

Jabriel828, 2022-09-21:
15bce158 3 net.c:184:mg_connect         1 -1 http://upload.qiniup.com
15bce159 3 net.c:184:mg_connect         2 -1 udp://8.8.8.8:53
15bce159 4 dns.c:251:mg_sendnsreq       1 resolving upload.qiniup.com @ 8.8.8.8, txnid 1
15bce159 3 sock.c:175:mg_send           2 0x3 0:0 35 err 0
15bce168 4 sock.c:665:mg_mgr_poll       2 r- tchrc
15bce168 3 sock.c:302:read_conn         2 0x3 snd 0/0 rcv 0/2048 n=275 err=0
15bce168 1 dns.c:152:dns_cb             Unexpected DNS response:
0000   00 01 81 80 00 01 00 0b 00 00 00 00 06 75 70 6c   .............upl
0010   6f 61 64 06 71 69 6e 69 75 70 03 63 6f 6d 00 00   oad.qiniup.com..
0020   01 00 01 c0 0c 00 05 00 01 00 00 01 4e 00 14 09   ............N...
0030   75 70 6c 6f 61 64 2d 7a 30 07 63 6c 6f 75 64 64   upload-z0.cloudd
0040   6e c0 1a c0 2f 00 05 00 01 00 00 02 23 00 10 0d   n.../.......#...
0050   75 70 6c 6f 61 64 2d 7a 30 2d 6f 73 73 c0 39 c0   upload-z0-oss.9.
0060   4f 00 05 00 01 00 00 00 d4 00 28 10 67 6c 6f 62   O.........(.glob
0070   61 6c 63 64 6e 64 79 6e 61 6d 69 63 05 71 69 6e   alcdndynamic.qin
0080   69 75 03 63 6f 6d 01 77 08 6b 75 6e 6c 75 6e 67   iu.com.w.kunlung
0090   72 c0 1a c0 6b 00 01 00 01 00 00 00 3c 00 04 b7   r...k.......<...
00a0   02 c1 ec c0 6b 00 01 00 01 00 00 00 3c 00 04 b7   ....k.......<...
00b0   02 c1 f0 c0 6b 00 01 00 01 00 00 00 3c 00 04 b7   ....k.......<...
00c0   02 c1 ed c0 6b 00 01 00 01 00 00 00 3c 00 04 b7   ....k.......<...
00d0   02 c1 ee c0 6b 00 01 00 01 00 00 00 3c 00 04 b7   ....k.......<...
00e0   02 c1 f1 c0 6b 00 01 00 01 00 00 00 3c 00 04 b7   ....k.......<...
00f0   02 c1 eb c0 6b 00 01 00 01 00 00 00 3c 00 04 b7   ....k.......<...
0100   02 c1 ea c0 6b 00 01 00 01 00 00 00 3c 00 04 b7   ....k.......<...
0110   02 c1 ef                                          ...             
15bce168 1 dns.c:185:dns_cb             stray DNS reply
15bce168 4 sock.c:665:mg_mgr_poll       1 -- tchRc
15bce19a 4 sock.c:665:mg_mgr_poll       2 -- tchrc

Jabriel828, 2022-09-21:
There are problems

scaprile, 2022-09-21:
I can confirm that using example code, the DNS response is not properly understood.

This host name is an alias of an alias of an alias, using any other shallower alias (e.g.: upload-z0.clouddn.com) resolves (see below)

> $ host upload.qiniup.com 8.8.8.8
Using domain server:
Name: 8.8.8.8
Address: 8.8.8.8#53
Aliases: 
upload.qiniup.com is an alias for upload-z0.clouddn.com.
upload-z0.clouddn.com is an alias for upload-z0-oss.clouddn.com.
upload-z0-oss.clouddn.com is an alias for globalcdndynamic.qiniu.com.w.kunlungr.com.
globalcdndynamic.qiniu.com.w.kunlungr.com has address 163.181.65.225
globalcdndynamic.qiniu.com.w.kunlungr.com has address 163.181.65.231
globalcdndynamic.qiniu.com.w.kunlungr.com has address 163.181.65.229
globalcdndynamic.qiniu.com.w.kunlungr.com has address 163.181.65.228
globalcdndynamic.qiniu.com.w.kunlungr.com has address 163.181.65.226
globalcdndynamic.qiniu.com.w.kunlungr.com has address 163.181.65.224
globalcdndynamic.qiniu.com.w.kunlungr.com has address 163.181.65.227
globalcdndynamic.qiniu.com.w.kunlungr.com has address 163.181.65.230

> http-client]$ ./example upload-z0.clouddn.com
170f9f7e 3 net.c:184:mg_connect         1 -1 upload-z0.clouddn.com
170f9f7e 3 net.c:184:mg_connect         2 -1 udp://8.8.8.8:53
170f9f7e 4 dns.c:251:mg_sendnsreq       1 resolving upload-z0.clouddn.com @ 8.8.8.8, txnid 1
170f9f7e 3 sock.c:175:mg_send           2 0x4 0:0 39 err 2
...
170fa03f 3 sock.c:302:read_conn         2 0x4 snd 0/0 rcv 0/2048 n=247 err=2
170fa03f 4 dns.c:155:dns_cb             globalcdndynamic.qiniu.com.w.kunlungr.com 1
170fa03f 3 dns.c:167:dns_cb             1 globalcdndynamic.qiniu.com.w.kunlungr.com is 163.181.65.225

Jabriel828, 2022-09-21:
I modified it and it can be parsed normally

This domain name is too special

if (mg_ntohs(h->num_answers) > 30) return 0;

closed 2022-09-21 by Jabriel828

jameshilliard, 2022-09-25:
> if (mg_ntohs(h->num_answers) > 30) return 0;

Hmm, coverity was flagging an issue with the `mg_ntohs` function [here](https://github.com/cesanta/mongoose/issues/1738#issuecomment-1256754427).