cesanta/mongoose #1763: Retagged 7.8 release?

gromgit, 2022-09-24:
Hi! I help up with the [Homebrew](https://brew.sh) project when I can, and I noticed that we updated our `mongoose` formula to v7.8 in the early hours of Aug 29 2022, shortly after your release. However, when I try to build it now, I get the following error:
```bash
$ brew install -s mongoose
==> Downloading https://github.com/cesanta/mongoose/archive/7.8.tar.gz
==> Downloading from https://codeload.github.com/cesanta/mongoose/tar.gz/refs/tags/7.8
##=O#- #                                                                      
Error: mongoose: SHA256 mismatch
Expected: 72cb3b5273fa969788d105da53cd4414274b702e761766eebcac9df66cfca2b8
  Actual: 55073dcd427ab9475731ad855e417884f4fbfb24b7d5694f6cabadbee1329f16
    File: /Users/aho/Library/Caches/Homebrew/downloads/0c4882ffc076f8afbcc01a65a85dbcc5ad1f9333bd82a46d749261a3202c352a--mongoose-7.8.tar.gz
```
i.e. the SHA256 hash of the current release tarball is no longer the same as the one from when v7.8 was first captured by our release process.

Was the 7.8 release tarball changed at some point? If so, may I know exactly what changed? Thanks much!

scaprile, 2022-09-24:
Does this address your question ?
https://github.com/cesanta/mongoose/releases/tag/7.8
There were no changes, to the git tag. Perhaps one of those links grabs from the master branch ?

gromgit, 2022-09-24:
Sorry, it doesn't. What I'm trying to point out is that: Homebrew's release process picked up a tagged 7.8 release around midnight on Aug 29, and proceeded to download the source and build precompiled binaries. The SHA256 checksum of that source tarball, as can be seen in [this commit](https://github.com/Homebrew/homebrew-core/commit/1919614907ca3d1f5b5522a2327f25624e330260), was `72cb3b5273fa969788d105da53cd4414274b702e761766eebcac9df66cfca2b8`.

The current 7.8 source tarball, _from the same URL_, now has a different SHA256 checksum:
```bash
$ curl -sL https://github.com/cesanta/mongoose/archive/7.8.tar.gz | sha256sum
55073dcd427ab9475731ad855e417884f4fbfb24b7d5694f6cabadbee1329f16  -
```

As Homebrew takes reproducible builds seriously, this is automatically flagged as an issue of concern, hence I'm asking if 7.8 was _re-released_ after its initial release that was picked up by Homebrew, and if so, what changed in the new 7.8 tarball?

cpq, 2022-09-24:
Yeah, I've moved the tag to include a small fix that was merged soon after 7.8 was first tagged.

Is there anything we could do other than cutting 7.9 ?

gromgit, 2022-09-25:
OK, since it's confirmed that:
1. the re-release was intended (as opposed to something sinister like a security compromise, and
2. the changes are actual code fixes rather than trivial documentation adjustments

I'll go ahead and update the `mongoose` formula's checksum, and bump the formula revision so that users will pick up the necessary changes. Thanks much for the quick explanation.!

carlocab, 2022-09-25:
For future reference, it'd be nice to avoid moving tags in the future. If you need to update a tag, just cut a new one (e.g. `7.8.1`). Moving tags is very disruptive for downstream packages and for anyone who has an existing clone of your repository before the tag was moved.

The [Git documentation](https://git-scm.com/docs/git-tag#_on_re_tagging) also has some pretty strong words about re-tagging.

cpq, 2022-09-28:
@gromgit @carlocab thanks!

closed 2022-09-28 by cpq