cesanta/mongoose #1768: Support for latest Zephyr

nomorecsgo, 2022-09-26:
- My goal is: We are planning on bringing up a product with Zephyr + Mongoose.
- My actions were: Building and running examples from examples/zephyr. Tried both in-tree and out-of-tree building for Zephyr 3.2.0.
- My expectation was: Successful building and running.
- The result I saw: Multiple issues with building and running. Zephyr seems to have changed include path but they are not updated in mongoose.h, also seeing Kconfig errors like: `error: MBEDTLS_DEBUG_LEVEL (defined at xxx/zephyrproject/zephyr/modules/mbedtls/Kconfig:99, modules\mbedtls\Kconfig:99) is        
assigned in a configuration file, but is not directly user-configurable (has no prompt). It gets its      
value indirectly from other symbols.`
While running http-server example on `nucleo_f746zg`, also observed `<err> net_sock: invalid access on sock 0 by thread 0x20019a28` etc, and the example seemed to have failed.
- My question is: Will there be updates to support latest Zephyr? There are features in the latest versions that we need to use.

scaprile, 2022-09-27:
The http-server builds with minimum changes to mongoose.h:
```
< #include <zephyr.h>
---
> #include "zephyr/kernel.h"

< #include <net/socket.h>
---
> #include <zephyr/net/socket.h>
```
No TLS errors, I just (locally) updated the docker image referenced by the example to the latest version of OS (3.2.0-rc3-57-gc4cc1621a5a2) and SDK (0.15.0)
Those `invalid access` errors are the result of Mongoose trying to use the network, and the network is not up. That is a Zephyr issue (probably an ST issue, F746 HAL Ethernet drivers are reluctant to come up). Of course a real life application should check the network is up.

scaprile, 2022-09-30:
We are migrating to Zephyr's official repo, there'll be updates soon

nomorecsgo, 2022-10-01:
@scaprile Thanks for the update. TLS error happens with the http-client example. I took a look and from [doc](https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_MBEDTLS_DEBUG_LEVEL) it seems that `CONFIG_MBEDTLS_DEBUG_LEVEL` can't be directly assigned and should be deferred from `CONFIG_MBEDTLS_LOG_LEVEL_DBG=y`, which makes `CONFIG_MBEDTLS_DEBUG_LEVEL=4`. The sample was built and ran on my nucleo board without issue after those changes. Yes it would be nice if all the zephyr samples can be merged into Zephyr repo and be built in-tree.

scaprile, 2022-10-01:
Sorry for the ambiguity, clarified the comment above.
The samples will also wait for the network to be up before launching Mongoose

scaprile, 2022-10-05:
#1783

closed 2022-10-05 by scaprile