cesanta/mongoose #1784: How to suppress warnings from OpenSSL

rafalsk, 2022-10-04:
# Environment

- mongoose version: 7.4

Hey there! Long time no hear, just two small questions;
1) any idea how to suppress warnings from OpenSSL? 
`ex. "SSL routines:ssl3_read_bytes:sslv3 alert certificate unknown:ssl\record\rec_layer_s3.c:1544:SSL alert number 46"`
Rationale: to save of CPU and synchronization usage (it all goes to console.. when many connections it can get messy quick).

2) lots of porting needed to 7.8 from 7.4?

And hey! Thanks a lot from Wizards. Great stuff.

We've integrated Mongoose heavily into GRIDNET OS (https://gridnet.org)... among many things the crypto-incentivized reverse CORS proxy apparatus is powered through it.. just take a look https://test.gridnet.org/wikipedia.org or  https://test.gridnet.org/google.com

cpq, 2022-10-05:
Great stuff, thanks!

1. That's not really a mongoose library question - more like an OpenSSL question
2. Not much - if you can be more specific, a more specific answer can be made

rafalsk, 2022-10-05:
Understood, closing, no more questions for now.

closed 2022-10-05 by rafalsk