cesanta/mongoose #1792: mg_sock_send run infinitely

tercelcn, 2022-10-09:
- My goal is: _describe_
- My actions were: _describe_
- My expectation was: _describe_
- The result I saw: _provide logs or description_
- My question is: mg_sock_send() run in an infinite loop if buffer length exceeds Maximum Segment Size(default 1440), send() return BSD_EWOULDBLOCK. I run mongoose in a thread which priority is osPriorityNormal, and net_core's priority also is osPriorityNormal, so that Thread net_core is always in the block state, because thread mongoose  is always running. 
# Environment

- mongoose version: 7.8 MG_ARCH_RTX
- Compiler/IDE and SDK: KEIL 5.37
- Target hardware/board:  stm32h743
- Connectivity chip/module: LAN8742a

cpq, 2022-11-09:
@tercelcn thank you!
Can you share a minimal example Keil project that has that issue, please?

cpq, 2022-11-24:
@tercelcn could you share your skeleton code please? We'd like to fix the issue.

cpq, 2022-12-19:
Closing this - no response. Please reopen if you could share a failing example with us.

closed 2022-12-19 by cpq