cesanta/mongoose #1799: What should Mongoose Server do when Websocket is unconventionally disconnected?

craigtao, 2022-10-17:
This does not occur in browsers (chrome, edge, IE), and when the browser is closed, the websocket disconnects normally.

But in .NET, when using webbrowser (based on the IE kernel), when the entire .NET application is closed, the js program inside webbrowser does not close websocket,  
This is actually the same principle as we directly unplug the network cable, which belongs to the abnormal disconnection of websocket.

At this point, if the mongoose server sends a message to the connection, the mongoose server crashes

craigtao, 2022-10-17:
I hope I had described this issue clearly

scaprile, 2022-10-17:
A WebSocket server must and should do what the respective RFC says it must and should do.
WebSocket runs on TCP and the server won't realize a client has disconnected until a timeout occurs, probably a keep-alive timeout. Once that happens, Mongoose will know the connection has closed and fire the proper event.
Of course crashing is not something a server should do, if you've found a scenario where that happens, please properly fill the issue template and provide enough information for the developers to reproduce it.

craigtao, 2022-10-17:
@scaprile  thanks scaprile.

You are right, my application scenario is: a thread A1 pushes state data to the web frontend through the websocket, and when the client is unconventionally disconnected, thread A1 uses the connection in the list to send data, and it crashes.

Or is it a flag **_is_closing_** or **_is_draining_** to determine whether the current connection is valid?

cpq, 2022-10-17:
@craigtao don't pass connection pointers to other threads, ever.

Make your thread just signal mongoose thread that data is there, and send data from mongoose thread.

craigtao, 2022-10-17:
@cpq thanks 
I'm sorry, there is a problem with my description.
I don't pass connection

**thread A1** is a **timer** mongoose provide

scaprile, 2022-10-17:
Are you checking your connection pointer is still valid ? Looks like you are dereferencing a NULL pointer
Things will be easier for both you and us if you would
- follow the issue template and properly describe your scenario (Mongoose version, OS, compiler, etc)
- provide minimum example code showing your point

As I've seen the word "thread" in your comments:
From [the docs](https://mongoose.ws/documentation/#2-minute-integration-guide)
> NOTE: Since Mongoose's core is not protected against concurrent accesses, make sure that all mg_* API functions are called from the same thread or RTOS task.

If you absolutely need multi-threading, take the time to go through the user guide (linked above) and [tutorials](https://mongoose.ws/tutorials/), there is one that shows you how to do [multi-threading](https://mongoose.ws/tutorials/multi-threaded/).

craigtao, 2022-10-18:
@scaprile  many thanks.

craigtao, 2022-10-18:
[image]
[image]

checking connection pointer is still valid, Is this okay? Screenshot above

but the issue remains,

It may be that my code to judge the connection is valid is incorrect?

GBY.

scaprile, 2022-10-18:
> Things will be easier for both you and us if you would
> 
>     * follow the issue template and properly describe your scenario (Mongoose version, OS, compiler, etc)
> 
>     * provide minimum example code showing your point

# Environment

- mongoose version: _X.Y_
- Compiler/IDE and SDK: ZZZ version _X.Y_
- Target hardware/board: _?_
- Connectivity chip/module: _?_-

Attach full example compilable code (minimum possible) where your issue can be reproduced. That means: someone here can compile your code and see what you claim

cpq, 2022-10-18:
@craigtao can you share your code? Please clean up it to the minimum that reproduces a problem.

cpq, 2022-11-09:
No response, closing

closed 2022-11-09 by cpq

scaprile, 2022-11-16:
A connection pointer (`c`) may have only two values: the address returned by your system's memory allocation function, or NULL. If it has a nonsense value, then either your system is returning that or you are trashing memory with runaway pointers, writing out of bounds, not following threading rules, etc.
Examples and tutorials check for `c == NULL`
https://mongoose.ws/tutorials/error-handling/#connection-retries