cesanta/mongoose #1804: OpenSSL configure TLS1.3 ciphers

xor-eax-eax, 2022-10-19:
When using mongoose as a https server with OpenSSL (here 1.1.1o) and TLS 1.3 it is not possible to restrict the allowed ciphers for TLS1.3. 
Actually the given string (for example "AES:...:!SHA:!aNULL:...") does not affect provided TLS1.3 ciphers by the webserver.
This may due to the usage of "SSL_CTX_set_cipher_list()", which only works for TLS <= 1.2, for TLS 1.3 openssl provides the function "SSL_CTX_set_ciphersuites()", see: https://www.openssl.org/docs/man1.1.1/man3/SSL_CTX_set_ciphersuites.html

Used nmap and sslscan are verify results.

Additionally it would be nice to be able to restrict / configure the allowed ECDHE groups used for key exchange.

cpq, 2022-11-19:
@xor-eax-eax  - please share your TLS init code. I suppose you're using the `ciphers` option there:

https://github.com/cesanta/mongoose/blob/76e6b23a4f0261dd534d33632f42ea1bba99dc85/mongoose.h#L1189

cpq, 2022-11-24:
@xor-eax-eax could you comment on the cipers settings please?

cpq, 2022-12-03:
No response, closing this.

closed 2022-12-03 by cpq