cesanta/mongoose #1806: How to control the priority order in the TLS cipher negotiation process by mongoose as https server

guangjung, 2022-10-20:
- My goal is: The mongoose https server may control the priority order in the TLS cipher negotiation process instead of https client. The best priority oder complies with the order  mongoose server configured (refer to struct mg_bind_opts bind_opts.ssl_cipher_suites).
- My actions were: 
step1.  add supported TLS1.2 ciphers, main modify code as follows
           struct mg_bind_opts bind_opts;
            ....
           #define SSL_CIPHER_SUITS "ECDHE-ECDSA-AES256-GCM-SHA384:ECDHE-ECDSA-AES128-GCM-SHA256:ECDHE-RSA-AES256-GCM-SHA384:ECDHE-RSA-AES128-GCM-SHA256:ECDHE-ECDSA-AES256-SHA384:ECDHE-ECDSA-AES128-SHA256:ECDHE-RSA-AES256-SHA384:ECDHE-RSA-AES128-SHA256"
            bind_opts.ssl_cipher_suites = SSL_CIPHER_SUITS
            ...
step2.   test negotiation TLS cipher  by openssl , example:
openssl s_client -connect 172.16.166.102:443 -tls1_2 -cipher "ECDHE-RSA-AES128-SHA256:ECDHE-RSA-AES256-SHA384:ECDHE-RSA-AES128-GCM-SHA256:ECDHE-RSA-AES256-GCM-SHA384"
- My expectation was: 
        the priority order in the TLS cipher negotiation process complies with the order I configured (refer to SSL_CIPHER_SUITS define).
- The result I saw: 
        the priority order in the TLS cipher negotiation process compiles with the https client (openssl s_client)
- My question is: 
How to control the priority order in the TLS cipher negotiation process by mongoose as https server? The best priority oder complies with the order  mongoose server configured (refer to struct mg_bind_opts bind_opts.ssl_cipher_suites).
# Environment

- mongoose version: 6.18/6.7/6.16
- Compiler/IDE and SDK: gcc (Ubuntu 16)
- Target hardware/board: N/A
- Connectivity chip/module: N/A

scaprile, 2022-10-20:
From the [openssl man pages](https://www.openssl.org/docs/man3.0/man1/openssl-s_client.html):
> _Although the server determines which ciphersuite is used it should take the first supported cipher in the list sent by the client_. 

Surely the TLS standard will give you more information.

guangjung, 2022-10-21:
Hi scaprile,
   I made some experiments on apache https server (also modified some TLS cipher configuration of apache) by the openssl tool, and found that  apache server took the first supported cipher in the server order list , and also the cipher was in client cipher list.
   Maybe the function could be strengthened in mongoose!
   Look forward to your reply, thanks!