cesanta/mongoose #1815: Error http-reverse-proxy example

roberto497, 2022-10-25:
I'm using the http-reverse-proxy example, and it's working with most of the s_backend_url I've tested, but it's not working with that specific url:
https://dprovaapi.procondutor.com.br:16443

In nginx it is working normally with the following configuration:

    location / 
        proxy_pass https://dprovaapi.procondutor.com.br:16443;
        proxy_ssl_server_name on;
    }

this is the log:

cc ../../mongoose.c -I../.. -W -Wall -DMG_IO_SIZE=128 -DMG_ENABLE_OPENSSL=1 -lssl -lcrypto  -o example main.c
./example
ab2c6e 3 mongoose.c:3462:mg_listen      1 0x4 http://0.0.0.0:8000
ab36de 3 mongoose.c:4370:accept_conn    2 0x5 accepted c0a80f11.47966 -> 0.8000
ab36e4 3 mongoose.c:4232:read_conn      2 0x5 snd 0/0 rcv 0/128 n=128 err=0
ab36e4 3 mongoose.c:4232:read_conn      2 0x5 snd 0/0 rcv 128/256 n=128 err=0
ab36e4 3 mongoose.c:4232:read_conn      2 0x5 snd 0/0 rcv 256/384 n=128 err=0
ab36e4 3 mongoose.c:4232:read_conn      2 0x5 snd 0/0 rcv 384/512 n=102 err=0
ab36e4 3 mongoose.c:3439:mg_connect     3 0xffffffffffffffff https://dprovaapi.procondutor.com.br:16443
ab36e4 3 mongoose.c:3439:mg_connect     4 0xffffffffffffffff udp://8.8.8.8:53
ab36e4 3 mongoose.c:4103:mg_send        4 0x6 0:0 46 err 0
ab36e7 3 mongoose.c:5274:mg_tls_init    3 Setting TLS, CA: ca.pem, cert: null, key: null
ab36eb 3 mongoose.c:5342:mg_tls_init    3 SSL client OK
ab36eb 3 main.c:35:forward_request      FORWARDING: GET /
ab36f0 3 mongoose.c:4232:read_conn      4 0x6 snd 0/0 rcv 0/128 n=62 err=0
ab36f0 3 mongoose.c:277:dns_cb          3 dprovaapi.procondutor.com.br is 187.72.254.106
ab36f0 3 mongoose.c:4314:mg_connect_res 3 0x7 -> bb48fe6a:16443 pend
ab36f7 3 mongoose.c:4370:accept_conn    5 0x8 accepted c0a80f11.47968 -> 0.8000
ab3717 1 mongoose.c:404:mg_error        3 0x7 socket error
ab3717 3 mongoose.c:3416:mg_close_conn  3 0x7 closed

scaprile, 2022-10-25:
Please sniff your network and contact the server admins.
Your desired host seems to accept only TLSv1.2 or a specific cipher suite:

```
No.     Time           Source                Destination           Protocol Length Info
     15 0.277615372    192.168.69.11         187.72.254.106        TLSv1    421    **Client Hello**

No.     Time           Source                Destination           Protocol Length Info
     17 0.315322808    187.72.254.106        192.168.69.11         TCP      62     16443 → 46572 [**RST**, ACK] Seq=1 Ack=366 Win=0 Len=0
```
It is rejecting the TLSv1 handshake but accepts a TLSv1.2 handshake

```
No.     Time           Source                Destination           Protocol Length Info
     28 31.595929012   192.168.69.11         187.72.254.106        TLSv1.2  573    Client Hello

No.     Time           Source                Destination           Protocol Length Info
     38 31.640442847   187.72.254.106        192.168.69.11         TLSv1.2  1030   Server Hello, Certificate, Certificate Status, Server Key Exchange, Server Hello Done
```

closed 2022-10-25 by scaprile

roberto497, 2022-10-26:
Exactly, our host has some peculiarities and does not work with all ciphers.
But it works with nginx and I did a test now and it also worked with libcurl. in my case I also need PKCS11 support, my question is can I adapt the reverse proxy using mongoose and libcurl?

Here's my code:

`
        ENGINE_load_dynamic();
	ENGINE_load_builtin_engines();
	e = ENGINE_by_id("dynamic");

	if (!e) {
		fprintf(stderr,"ENIGNE_by_id failure\n");
		exit(-1);
	}

	if(!ENGINE_ctrl_cmd_string(e, "SO_PATH", ENGINE_PKCS11_PATH, 0)||
	   !ENGINE_ctrl_cmd_string(e, "ID", "pkcs11", 0) ||
	   !ENGINE_ctrl_cmd_string(e, "LIST_ADD", "1", 0) ||
	   !ENGINE_ctrl_cmd_string(e, "LOAD", NULL, 0) ||
	   !ENGINE_ctrl_cmd_string(e, "MODULE_PATH", p11lib, 0) ||
	   !ENGINE_ctrl_cmd_string(e, "PIN", pin, 0) ) {
			fprintf(stderr,"ENGINE_ctrl_cmd_string failure\n");
			exit(-1);
	}

	if (!ENGINE_init(e)) {
		fprintf(stderr,"ENGINE_init failure\n");
		exit(-1);
	}

	curl_global_init(CURL_GLOBAL_ALL);
	curl = curl_easy_init();
	if (!curl) {
		fprintf(stderr,"couldn't init curl\n");
		exit(-1);
	}

	curl_easy_setopt(curl, CURLOPT_SSLCERTTYPE, "PEM");
	curl_easy_setopt(curl, CURLOPT_SSLCERT, "/home/roberto/procondutor.crt");
	curl_easy_setopt(curl, CURLOPT_SSLKEY, "pkcs11:model=19C43A06010D0000;serial=022A00100020FA04;token=TOKEN;object=Procondutor;type=private");
	curl_easy_setopt(curl, CURLOPT_VERBOSE, 1);
	curl_easy_setopt(curl, CURLOPT_USE_SSL, CURLUSESSL_ALL);
	curl_easy_setopt(curl, CURLOPT_SSL_VERIFYPEER, 0);
	curl_easy_setopt(curl, CURLOPT_SSL_VERIFYHOST, 0);
	curl_easy_setopt(curl, CURLOPT_SSLENGINE, "pkcs11");
	curl_easy_setopt(curl, CURLOPT_SSLKEYTYPE, "ENG");
	curl_easy_setopt(curl, CURLOPT_SSLENGINE_DEFAULT, 1L);
	curl_easy_setopt(curl, CURLOPT_WRITEDATA, NULL);
	curl_easy_setopt(curl, CURLOPT_WRITEFUNCTION, write_data);

	CURLcode code;

	code = CURLE_COULDNT_CONNECT;
	curl_easy_setopt(curl, CURLOPT_URL, url);
	curl_easy_setopt(curl, CURLOPT_VERBOSE, 0L);
	code = curl_easy_perform(curl);
`

**My idea is to use mongoose's http-server + libcurl to terminate reverse proxy**

roberto497, 2022-10-28:
I found the problem, as my host has WAF, I need to implement SNI as below:

SSL_set_tlsext_host_name(tls->ssl, host);