cesanta/mongoose #1827: mqtt empty payload

Stifael, 2022-11-02:
- My goal is: subscribe to a topic which has been deleted on the broker
- My expectation was: successful mqtt subscription with zero-byte payload
- The result I saw: 
   - client disconnects from broker because of the function **mg_mqtt_parse**, which considers a zero-byte message as malformed (https://github.com/cesanta/mongoose/blob/master/mongoose.c#L3198) 
- My question is: how to handle subscription of a deleted message with the cesanta-mongoose library

# Environment

- mongoose version: 7.8
- Compiler: gcc (Debian 10.2.1-6) 10.2.1 20210110
- Target hardware/board: amd64

scaprile, 2022-11-02:
Topics do not live on the broker, you either publish to a topic or not and you can subscribe to whatever, you just won't receive anything if nobody publishes there, UNLESS your broker does not allow certain topics because of internal policies and such. The response to a subscription is a SUBACK, not an empty message. Is that empty message coming from a retained message on that topic ? Did you sniff your traffic ?
Proof: take the mqtt-client example, modify the subscription topic to some long nonsense, run it: it subscribes.
So, can you submit a working (minimal) example that proves your point (a network capture will also help), so we can check what is going on ?

Stifael, 2022-11-03:
It seems my explanation was not very clear.
The order of execution is as follow:
- client_1 subscribes to a topic_x, which was published with retain flag to true
- the retained messate of topic_x gets deleted. this can be accomplished directly on the broker or sending a message of zero-payload to topic_x (see https://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages/#:~:text=There%20is%20also%20a%20very,retained%20message%20for%20that%20topic.)
- after the retained message has been deleted, the broker still sends that messate with zero-payload to client_1. this is expected
- client_1 receives that message with zero-payload, but mongoose considers such a messate as malformed (see (https://github.com/cesanta/mongoose/blob/master/mongoose.c#L3198)

I uploaded an example [here](https://github.com/Stifael/cesanta-mongoose-mqtt-test).
It is an example taken from mongoose-examples, but after a susccessful connection to the broker, the client sends a zero-payload message to the same topic for which the client has previously subscribed to. the broker used in the example is a mosquitto broker.

scaprile, 2022-11-03:
Thanks, that is clear and seems reproducible. To "delete" the retained message you published a new message with zero payload and the retain flag set.
Did you try a clean session start ?
We'll try to reproduce it and check.

PR #1831 (2022-11-03): Accept zero-length payload messages
Zero-length payload messages are valid messages, they can be sent with regular clients (mosquitto_pub -m "" or -n) and are fond to occur when a producer "deletes" a retained message and the consumer does not start with a clean session.

resolves #1827
closed 2022-11-03 by cpq