cesanta/mongoose #1828: GCC 12 on ARM: 'mg_sha1_transform' reading 64 bytes from a region of size 0 [-Werror=stringop-overread]

jcorporation, 2022-11-02:
- My goal is: Compile mongoose with gcc 12.2.1
- My actions were: normal build
- My expectation was: build successful
- The result I saw: compiler error in function mg_sha1_transform
- My question is: compiler or mongoose failure?

# Environment

- mongoose version: 7.8
- Compiler/IDE and SDK: GNU 12.2.1
- Target hardware/board: armv7l:armv6l:armv5tel

Build log:
```
[   34s] In function 'mg_sha1_update',
[   34s]     inlined from 'mg_sha1_final' at /home/abuild/rpmbuild/BUILD/mympd-10.1.0/dist/mongoose/mongoose.c:3847:3:
[   34s] /home/abuild/rpmbuild/BUILD/mympd-10.1.0/dist/mongoose/mongoose.c:3829:7: error: 'mg_sha1_transform' reading 64 bytes from a region of size 0 [-Werror=stringop-overread]
[   34s]  3829 |       mg_sha1_transform(context->state, &data[i]);
[   34s]       |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[   34s] /home/abuild/rpmbuild/BUILD/mympd-10.1.0/dist/mongoose/mongoose.c:3829:7: note: referencing argument 2 of type 'const unsigned char[64]'
[   34s] /home/abuild/rpmbuild/BUILD/mympd-10.1.0/dist/mongoose/mongoose.c: In function 'mg_sha1_final':
[   34s] /home/abuild/rpmbuild/BUILD/mympd-10.1.0/dist/mongoose/mongoose.c:3701:13: note: in a call to function 'mg_sha1_transform'
[   34s]  3701 | static void mg_sha1_transform(uint32_t state[5],
[   34s]       |             ^~~~~~~~~~~~~~~~~
[   34s] In function 'mg_sha1_update',
[   34s]     inlined from 'mg_sha1_final' at /home/abuild/rpmbuild/BUILD/mympd-10.1.0/dist/mongoose/mongoose.c:3850:5:
[   34s] /home/abuild/rpmbuild/BUILD/mympd-10.1.0/dist/mongoose/mongoose.c:3829:7: error: 'mg_sha1_transform' reading 64 bytes from a region of size 0 [-Werror=stringop-overread]
[   34s]  3829 |       mg_sha1_transform(context->state, &data[i]);
[   34s]       |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[   34s] /home/abuild/rpmbuild/BUILD/mympd-10.1.0/dist/mongoose/mongoose.c:3829:7: note: referencing argument 2 of type 'const unsigned char[64]'
[   34s] /home/abuild/rpmbuild/BUILD/mympd-10.1.0/dist/mongoose/mongoose.c: In function 'mg_sha1_final':
[   34s] /home/abuild/rpmbuild/BUILD/mympd-10.1.0/dist/mongoose/mongoose.c:3701:13: note: in a call to function 'mg_sha1_transform'
[   34s]  3701 | static void mg_sha1_transform(uint32_t state[5],
[   34s]       |             ^~~~~~~~~~~~~~~~~
[   36s] cc1: all warnings being treated as errors
```

scaprile, 2022-11-02:
Since Mongoose does not compile as a standalone, please tell us exactly where are you building, what are you building, and how are you building it, so we can reproduce it here and check what triggers that.

jcorporation, 2022-11-03:
I am unsure why, but I could not reproduce this issue today. Sorry for the noise!

closed 2022-11-03 by jcorporation