cesanta/mongoose #1852: Misleading documentation

TortillaZHawaii, 2022-11-11:
Hi, it looks like one of the tutorials https://mongoose.ws/tutorials/json-rpc-over-websocket/ references [JSON-RPC 2.0](https://www.jsonrpc.org/specification) specification, but mongoose doesn't fully support it.

Especially it doesn't return `"jsonrpc": "2.0"` to the Response object.

It is extremely misleading, since JSON-RPC 2.0 was created more than 10 years ago and parts of it are supported like named parameters in responses.

TortillaZHawaii, 2022-11-12:
It looks to me like all of the JSON-RPC 2.0 is implemented, apart from the mentioned `"jsonrpc": "2.0"`. Here is a list of differences between [JSON-RPC 1.0 vs JSON-RPC 2.0](http://www.simple-is-better.org/rpc/#differences-between-1-0-and-2-0).

cpq, 2022-11-13:
Nothing stops you adding that in the response. Mongoose provides a way to do it, that's all.

closed 2022-11-13 by cpq