cesanta/mongoose #1855: mg_timer It is not triggered according to the expected set period duration

craigtao, 2022-11-14:
- My goal is: _describe_
- My actions were: _describe_
- My expectation was: _describe_
- The result I saw: 
    **mg_mgr_poll(&mD->manager, 5);**
    **mg_timer_init(&timerPush, 1, MG_TIMER_REPEAT, &Private::onTablePush, this);**
    timerPush doesn't actually have 10 millisecond, according to my observation it actually fires once in 1 second
- My question is: _your question_

# Environment

- mongoose version: 7.6
- Compiler/IDE and SDK: gcc 7.1.0
- Target hardware/board: intel x86 i5
- Connectivity chip/module: _?_

scaprile, 2022-11-14:
It is hard to tell with such a short and non-standard piece of code. Did you try any of the examples ? They do work as expected and show you how to do things properly, The manager must be polled within an infinite loop, the timer must be init only once, and you are wanting a 1ms period with that function call. Your manager timeout must be less than your requested timer period.
I suggest you follow this tutorial: http://mongoose.ws/tutorials/timers/

closed 2022-11-14 by scaprile

craigtao, 2022-11-14:
Hi, thanks your reply.

so sorry a lot of code details can't be disclosed, I can't provide what you think is a "non-standard piece of code", I can't provide code that you can run directly.

I tried to provide a code snippet that indicates the current logic

[image]

craigtao, 2022-11-14:
onTablePush function, I didn't do any processing, just execute the printf function

craigtao, 2022-11-14:
I found a previous issue, is the issue I am currently experiencing related to it?

fixed poll timeout being larger than timer period #1587

scaprile, 2022-11-14:
You are providing a manager timeout of 10ms and expect a timer to fire every 1ms. That won't happen, as explained in the tutorial.
Besides that, you claim "1 second", what is your time base ? What is your OS ? Mongoose may be defaulting to time() and it may be providing 1s resolution.
Please try the examples, those do work: https://github.com/cesanta/mongoose/tree/master/examples/timers

scaprile, 2022-11-14:
> I found a previous issue, is the issue I am currently experiencing related to it?
> 
> fixed poll timeout being larger than timer period #1587

No, that is a fix to an example, but if you look closely you are committing the same mistake I just pointed out to you

craigtao, 2022-11-14:
> You are providing a manager timeout of 10ms and expect a timer to fire every 1ms. That won't happen, as explained in the tutorial. Besides that, you claim "1 second", what is your time base ? What [...]

I have update the picture, manager timeout is 10ms, timer is 20ms,

OS is Lynx, it's a realtime os,

and, I have read the manual instruction about mg_timer

craigtao, 2022-11-14:
mongoose official document

craigtao, 2022-11-14:
maybe the time accuracy of mg_timer only ≥1s? mongoose 7.6 version

scaprile, 2022-11-14:
> Besides that, you claim "1 second", what is your time base ? What is your OS ? Mongoose may be defaulting to time() and it may be providing 1s resolution. Please try the examples, those do work: https://github.com/cesanta/mongoose/tree/master/examples/timers

Mongoose doesn't know about Lynx, `mg_millis()` tries to guess and defaults to `time()`, as already told. If your `time()` has a 1s resolution, then Mongoose can't do magic.
https://github.com/cesanta/mongoose/blob/1b82aa02aa3d6fecfc23ed0a94c6917f139de1ad/mongoose.c#L4634-L4667

Run an example and see how it works, if an example does not work then your problem is on your time base, modify the source code appropriately or use 7.7/7.8 where you can define `MG_ENABLE_CUSTOM_MILLIS` and provide your own implementation of `mg_millis()`.