cesanta/mongoose #1902: cannot free string returned by mg_strdup()

nirodho, 2022-12-05:
- My goal is: _describe_
- My actions were: _describe_
- My expectation was: _describe_
- The result I saw: _provide logs or description_
- My question is: _your question_

# Environment

- mongoose version: _X.Y_
- Compiler/IDE and SDK: ZZZ version _X.Y_
- Target hardware/board: _?_
- Connectivity chip/module: _?_
- Target RTOS/OS (if applicable): _?_

scaprile, 2022-12-05:
`free((void *) str2.ptr)`... `free((char *) str2.ptr)`... just get rid of the `const` as your compiler is instructing you to do.

closed 2022-12-05 by scaprile

nirodho, 2022-12-05:
Thanks. I thought there might be some reason not to cast away the const
modifier.

I just started to use Mongoose to interact with the ESP32 board, by the
way. I'm loving it :)

Thanks again.

Jack

PR #1903 (2022-12-05): Update README.md
closes #1902