cesanta/mongoose #1930: [Win32] loopback issue

gvanem, 2022-12-20:
The sample-program `example/mip-pcap/main.c` does not work well as a HTTP-server
on Windows (using WinPcap v. 4.1.0.2980).

But the DHCP feature works fine. So when I got a `10.0.0.47` assigned from my router, I tried
a `http://10.0.0.47:8000` in Google Chrome. With the Developer-Console active, this is what I got:
```
GET / HTTP/1.1
Host: 10.0.0.147:8000
Connection: keep-alive
Pragma: no-cache
Cache-Control: no-cache
DNT: 1
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/108.0.0.0 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
Accept-Encoding: gzip, deflate
Accept-Language: en-GB,en;q=0.9,nb;q=0.8
```

The sent packets from Chrome are looped back to Chrome!

Further the function `pcap_setdirection(ph, PCAP_D_IN);` is called w/o checking the return value.
It's not supported on Windows.

AFAICS, the fix would be to call `PacketSetLoopbackBehavior(pw->adapter, NPF_DISABLE_LOOPBACK)`. 
Or call `pcap_open (... PCAP_OPENFLAG_NOCAPTURE_LOCAL,...);` instead.

scaprile, 2022-12-20:
Well, this example was developed on Windows by a collaborator, can you please run Wireshark on your interface and post a capture file ? Also, please, describe your networking scenario.

gvanem, 2022-12-20:
I used tcpdump on the side; [here](https://github.com/cesanta/mongoose/files/10269243/port-8000-pcap.zip) it is. 

I'm running both `mip-pcap-example.exe` and Chrome on the same machine.

scaprile, 2022-12-20:
But you said you have  a router, who assigns IP addresses ? How does your board reach that ? Is there a virtual bridge ? Which interface do you connect to ? loopback ? Ethernet ? WiFi ?
Loopback interfaces (at least) in Windows loop traffic back, in such a case, did you try connecting to other interface ?

SkateScout, 2022-12-20:
Hi, i checked the pcap file. The server is running fine. There is no error.
The response is HTTP Code 200 with Content-Length (Little bit Ugly).
And Without Content-Type the payload is simply an mirror of the request.
This is an legal answer for an demo.

GET / HTTP/1.1
Host: 10.0.0.147:8000
Connection: keep-alive
Cache-Control: max-age=0
DNT: 1
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/108.0.0.0 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
Accept-Encoding: gzip, deflate
Accept-Language: en-GB,en;q=0.9,nb;q=0.8

HTTP/1.1 200 OK
Content-Length: 0000000475

GET / HTTP/1.1
Host: 10.0.0.147:8000
Connection: keep-alive
Cache-Control: max-age=0
DNT: 1
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/108.0.0.0 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
Accept-Encoding: gzip, deflate
Accept-Language: en-GB,en;q=0.9,nb;q=0.8

gvanem, 2022-12-21:
>  How does your board reach that ? Is there a virtual bridge ? Which interface do you connect to ? loopback ? Ethernet ? WiFi ?

No bridge or loopback. My setup is:
```
______________                         ______________
|   Router    |                        |   Win-10    |
| 10.0.0.1    |------------Ethernet--- |  10.0.0.10  |
|_____________|                        |_____________|
```

Running the MIP-example and Chrome on the same Win-10 PC.

SkateScout, 2022-12-21:
You can check the pcap with wireshark, the packets are not simply looped back to chrome.  Since the "200 OK\n\rContent-Length: ..." was added. This show that there was an http server in the communication as expected. On the other hands how should the packets come back without an endpoint. pcap capture/inject the packets but to get an TCP connection established there is an syn->syn/ack logik to be managed. And chrome would not accept the incomming connection.

scaprile, 2022-12-21:
There is no loopback, there is always a single Ethernet frame containing a TCP segment for every action.
See the event handler:
https://github.com/cesanta/mongoose/blob/21d3678fa7afd9f0f63eecc77b35b3f65b66f35f/examples/mip-pcap/main.c#L58-L79
If you don't request any of those URIs then the handler defaults to the action in line 74
https://github.com/cesanta/mongoose/blob/21d3678fa7afd9f0f63eecc77b35b3f65b66f35f/examples/mip-pcap/main.c#L74-L75
and that, as you can see, sends you back your request in the response body

closed 2022-12-21 by scaprile