cesanta/mongoose #1940: Teensy 4.1 support ?

psykokwak-com, 2022-12-24:
- My goal is: use this library with a teensy 4.1 board
- My actions were: put mongoose.c|h on my arduino project
- My expectation was: build succesful
- The result I saw: build error (see log below)
- My question is: how to use mongoose library with teensy 4.1 board ?

# Environment

- mongoose version: latest
- Compiler/IDE and SDK:  Teensyduino, Version 1.57
- Target hardware/board: Teensy 4.1
- Connectivity chip/module: native ethernet
- Target RTOS/OS (if applicable): I don't know.

# Build Log

> mongoose.c:20: In file included from
> mongoose.h: 65:51: fatal error: mongoose_custom.h: No such file or directory

scaprile, 2022-12-24:
You need to provide a suitable environment, "native ethernet" is not a connectivity chipset nor a module, your hardware has an Ethernet chip and either your RTOS/OS (which you must know to make this work) provides a Berkeley-compatible sockets environment or Mongoose needs a driver for your chip.
https://www.pjrc.com/store/teensy41.html
I see a photo of an MiMX-RT1062 and the docs say it has a  DP83825 PHY.
We currently don't support that chipset, so your options depend on your OS or writing a driver
Please read the documentation on porting, the second links explains why you get that compilation error
https://mongoose.ws/documentation/
https://mongoose.ws/documentation/#custom-build
https://github.com/cesanta/mongoose/tree/master/examples/arduino/w5500/
and the last link is an example of using Arduino with a WizNet W5500 in case you want to explore that path.

closed 2022-12-24 by scaprile

psykokwak-com, 2023-01-02:
Thanks for these informations.
As there is a lwip based network library ( https://github.com/ssilverman/QNEthernet ), do you think its possible to make mongoose working on it ?

scaprile, 2023-01-02:
Mongoose already works with lwIP.
Making it work with that library is other business

psykokwak-com, 2023-01-02:
Not with that library but with the underlying lwip ?
How to make a custom build with lwip ?

scaprile, 2023-01-02:
Please clone this repo, grep for "lwip" and see the dozens of examples.

psykokwak-com, 2023-01-06:
I'm trying to build it with LWIP but my LWIP stack uses "NO_SYS=1".
And with "NO_SYS=1" I cannot enable LWIP_SOCKET.
Is LWIP_SOCKET necessary for mongoose ?

scaprile, 2023-01-06:
Yes, Mongoose dropped support for the RAW API long ago
If you can get an API with direct access to send and receive Ethernet frames, you can use its built-in TCP/IP stack: MIP. You'll have to write a driver for that

psykokwak-com, 2023-01-07:
Ok. I started to write a mip driver for teensy41.
Is there a way to disable all mip checksums calculation (for ip/tcp/udp/icmp) to let the hardware to do that ?

scaprile, 2023-01-07:
We are currently not focusing on those performance improvements since with today MCUs that is barely noticeable (if at all) in typical applications.

This section of this tutorial explains the main parts of a driver https://mongoose.ws/tutorials/stm32/nucleo-f429zi-usb-rndis/#under-the-hood

psykokwak-com, 2023-01-07:
Thanks.
I finished writing the driver and it works.
How to publish it?

scaprile, 2023-01-08:
Cool! 
Just send a PR (Pull Request), you'll be asked to sign a form if it is considered for inclusion. Thanks.

scaprile, 2023-01-08:
#1954