cesanta/mongoose #1945: failed: Error during WebSocket handshake: Sent non-empty 'Sec-WebSocket-Protocol' header but no response was received

mansamtsui, 2022-12-31:
hello
i use this lib in webgl platform（pc browers chrome） ，call mg_ws_connect function to connect server got this error :
failed: Error during WebSocket handshake: Sent non-empty 'Sec-WebSocket-Protocol' header but no response was received

i write code simple like example ：
mg_ws_connect(&mgr, s_url, fn, &done, "%s", "Sec-WebSocket-Protocol: echo\r\n");

use the same code in pc platform(win 32 application)  is work, connected.

i want to set empty  'Sec-WebSocket-Protocol'  in websocket connect request, but checked all example , but failed

please help!!

scaprile, 2022-12-31:
You should start by studying examples and tutorials relevant to what you are trying to do https://mongoose.ws/tutorials/websocket-server/ https://mongoose.ws/tutorials/websocket-client/ , then reading the documentation https://mongoose.ws/documentation/
Then, if you have a problem, sniff your network to find the reasons, and finally, if you have an issue, properly fill the issue template and provide us with enough information to help you, like the smallest code reproducing your issue that we can run here.
Your description is quite vague and honestly I can't grasp anything of what you are trying to do. Do you have an event manager running ? A connection event handler callback ?

mansamtsui, 2023-01-03:
First at all thanks for your reply, and i am sorry for my described the problem too briefly.
I have read the tutorials and documentation that your wrote ,  but also have this issue, and i also searched it in anthor closed issue of this repository, and i still have no idea how to resolve it.

this is my test code that is in the tutorials

I declare these variables globally

     struct mg_mgr mgr; 
    bool done = false;
    struct mg_connection* c

in my application, i write this in Initialise function.

    mg_mgr_init(&mgr);        // Initialise event manager
    mg_log_set(MG_LL_DEBUG);  // Set log level
    c = mg_ws_connect(&mgr, s_url, fn, &done, NULL);     // Create client

and i write poll function in my main cycle function to polling the connection finish 

     while (GetMessage(&msg, nullptr, 0, 0))
    {
        mg_mgr_poll(&mgr, 1000);  // Wait for echo
    }

and fn function is the event handler callback that my declare

   static void fn(struct mg_connection* c, int ev, void* ev_data, void* fn_data) {
       if (ev == MG_EV_OPEN) {
         //connecting, No exception received
      }
     else if (ev == MG_EV_RESOLVE )
      {
         //The above exception occurs when the message is received
      }
      else if (ev == MG_EV_WS_OPEN) {
         //connected, Call upper level functions
     }
  }

All the above codes are normal when I use visual studio to write win32 applications on the window platform, and can successfully connect to server programs written in java.

However, when I use the webgl platform web application, the above exception occurs

First, I use the unity engine to develop programs. It compiles c code into webassembly and runs it on the browser through the compiler

In the above fn function, the change process of ev parameter I received is 
MG_ EV_ OPEN -- MG_ EV_ RESOLVE --- MG_ EV_ CONNECT --- MG_ EV_ CLOSE
and finaly i got the exception in brower debug console window. Before the exception is printed, the ev parameter value is  MG_ EV_ RESOLVE

The java server side shows that the connection I sent is automatically disconnected immediately after it is successfully connected, and there is no exception on the java side

I'm sorry for the above description. I hope it can give you enough information

mansamtsui, 2023-01-03:
log：​ 
0 3 mongoose.c:4330:mg_connect_resolved 1 0x5 -> a120f5d:9334 pend

​  WebSocket connection to 'ws://10.18.15.93:9334/' failed: Error during WebSocket handshake: Sent non-empty 'Sec-WebSocket-Protocol' header but no response was received
createPeer @ build_webgl.framework.js.gz:2

Before the following exception reports, there is a log in this line：0 3 mongoose.c:4330:mg_connect_resolved 1 0x5 -> a120f5d:9334 pend

mansamtsui, 2023-01-03:
My java server got this request header information of client ：

GET / HTTP/1.1
Host: 10.17.1.124:9334
Connection: Upgrade
Pragma: no-cache
Cache-Control: no-cache
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/108.0.0.0 Safari/537.36 Edg/108.0.1462.54
Upgrade: websocket
Origin: http://127.0.0.1:8080
Sec-WebSocket-Version: 13
Accept-Encoding: gzip, deflate
Accept-Language: zh-CN,zh;q=0.9,en;q=0.8,en-GB;q=0.7,en-US;q=0.6
Sec-WebSocket-Key: rOOOUFi5fDxgA9CmyPildQ==
Sec-WebSocket-Extensions: permessage-deflate; client_max_window_bits
Sec-WebSocket-Protocol: binary

we can see “Sec-WebSocket-Protocol”  exactly not empty in websocket connection request， but in above test code I did empty the fifth parameter of the function

scaprile, 2023-01-03:
> Then, if you have a problem, **sniff your network to find the reasons**, and finally, if you have an issue, **properly fill the issue template** and provide us with enough information to help you, like the **smallest code reproducing your issue** that we can run here.

and a network capture file to see the traffic where you think you've found something that is not in accordance to [RFC6455](https://www.rfc-editor.org/rfc/rfc6455)

That is:
- Fill the issue template, it exists for a reason: to save yours and our time
- Sniff your network and provide a pcap file where the issue can be seen, please clearly explain where you see something that is not RFC compliant
- Provide the smallest piece of code where we can reproduce your issue, in order to fix it.

mansamtsui, 2023-01-03:
OK, maybe everyone has different communication habits. I'll find out the problem myself, thank you

closed 2023-01-03 by scaprile