cesanta/mongoose #1953: Help debugging/solving issue with an http-server

diamante0018, 2023-01-07:
- My goal is: Create a simple HTTP server
- My actions were: Base my code from the example web page https://mongoose.ws/tutorials/http-server/
- My expectation was: To serve files using a TCP socket via Mongoose API (our code can be viewed here [Download.cpp](https://github.com/XLabsProject/iw4x-client/blob/develop/src/Components/Modules/Download.cpp)
- The result I saw: After around 6 hours of uptime, the server suddenly becomes unreachable (after having worked fine) to HTTP requests. The handler only receives `MG_EV_POLL` but never `MG_EV_HTTP_MSG` again.
- My question is: What could I do to investigate this issue and solve it?

## Extra info
We also have a UDP socket running on the same port. Our application has used Mongoose like this for many years with no issues. We did update recently from a 2019 commit of Mongoose all the way to 7.8
I call `mg_http_listen` on the "main" thread of my application, and then I start a separate thread just for Mongoose that calls `mg_mgr_poll(&mgr, 1000);`
It works fine for about 6 hours, then the EventHandler, which is my function with the URI matching code that does some different actions based on the URI, stops receiving any data. We have been able to get this issue on different machines compiled with different versions of MSVC, so we are likely misusing Mongoose API or maybe it is something else.
I have attached a debugger to my application and discovered that `mg_mgr_poll` is called as expected every 1000 seconds and that the event handler is triggered only for `MG_EV_POLL` EV types after this breakage occurs so it seems that the Async thread we have just for Mongoose isn't stuck at all and neither is the "Event Handler" function which has a
`cpp
if (ev != MG_EV_HTTP_MSG)
{
	return;
}
`
at the beginning.

I would like to know if there is something I can do to better debug this issue once it occurs and to find out why HTTP requests are ignored. Thanks.

# Environment

- mongoose version: 7.8 also got the same results on the latest 'master'
- Compiler/IDE and SDK: VS2022 version 17.4.3 Win10 DSK
- Target hardware/board: Win32
- Connectivity chip/module: _?_
- Target RTOS/OS (if applicable): _?_

scaprile, 2023-01-07:
Well, your code is a bit far too long for a lazy guy like me to read it all.
I did a quick search for the word "thread" and found it 26 times, so I must paste a quick reminder:

From [the docs](https://mongoose.ws/documentation/#2-minute-integration-guide)
> NOTE: Since Mongoose's core is not protected against concurrent accesses, make sure that all mg_* API functions are called from the same thread or RTOS task.

If you absolutely need multi-threading, take the time to go through the user guide (linked above) and [tutorials](https://mongoose.ws/tutorials/), there is one that shows you how to do [multi-threading](https://mongoose.ws/tutorials/multi-threaded/).

Beyond that, the event manager sleeps for the time you set only if nothing arrives on a socket, so if you always see the same time, nothing is coming from your sockets. If you are not trying to do multi-threading, then check here:

https://github.com/cesanta/mongoose/blob/042465387f941db3331f14af99ef61d8444b57a3/src/sock.c#L494-L619

Check which compile options you are using, and please try to shrink your code to the minimum piece that reproduces your problem.

diamante0018, 2023-01-07:
No, my code shouldn't be trying to do any multithreading.
I just start a separate thread that calls mgr_poll, and that's about it. No other thread interacts with the API besides that one.

`MG_ARCH` is set to 2 so only this code is executed
https://github.com/cesanta/mongoose/blob/master/src/sock.c#L579
Same for the mg_iotest test found in mongoose.c file.

I will try to produce a small code sample but it might take me some time.
For now I can point to this line of code where the "event handler" bound with `mg_http_listen` starts.
https://github.com/XLabsProject/iw4x-client/blob/develop/src/Components/Modules/Download.cpp#L644

Thanks for the reply

scaprile, 2023-01-07:
If you follow down from line 579 you'll arrive to the `select()`call, Mongoose blocks there and returns when either the OS tells there is some action pending on a socket (ready to transmit, data to read) or the timeout elapses. You should let a server running debuggable code and when you found a problem pause it and step it throug that section to see what happens with your sockets, either Mongoose closed them or your OS did. You can also enable debugging and check the logs: https://mongoose.ws/documentation/#mg_log_set

I don't get your "using a TCP socket and a UDP socket on the same port", you don't open the socket, Mongoose does, and TCP and UDP port numbers belong to different realms, like 538 E 1st street in Manhattan and 538 E 1st street in Boston.
BUT, you need different calls, one to `mg_http_listen()` and the other to `mg_listen()` 
Check [this tutorial](https://mongoose.ws/tutorials/captive-dns-portal/) and the [API docs](https://mongoose.ws/documentation/)

cpq, 2023-01-08:
@diamante0018 please read https://mongoose.ws/documentation/#best-practices , especially the note marked red.

Long story short: use `mg_http_reply()` or chunked encoding, `mg_http_...chunk()`. The set Content-Length AND `c->is_resp`. If you use `mg_printf`/`mg_send`, don't forget to set Content-Length (which you do), and `c->is_resp` (which you do not):

https://github.com/XLabsProject/iw4x-client/blob/90b124420ff6f2803b317f7a7865d4135d86d8b1/src/Components/Modules/Download.cpp#L636

 Otherwise, your connections are not marked as "done" - and apparently are piling up, eventually filling up the `FD_SETSIZE` limit. Take a look at the number of opened connections when your server stops responding (you can probably trigger that fast by bombarding your server with requests on c->is_resp unfriendly URIs). I anticipate your server having a lot of lingering connections.

diamante0018, 2023-01-08:
Thank you @cpq. As I suspected, it was most certainly me misusing the Mongoose API.
It is unfortunate I was not using mg_http_serve_dir. I am confident it will fix my issue (I will close this thread as soon as I confirm).
Unfortunately, I did not write the code originally in the application  I'm maintaining. As I stumbled upon it earlier this year, it did not occur to me t use mg_http_serve_dir or was too scared to change very old code. So I got a few users complaining the server 'stopped responding' and didn't know what to do.

cpq, 2023-01-08:
@diamante0018 you're welcome.
And hey, it's weekend. Time to break stuff!

diamante0018, 2023-01-09:
Seems like it's fixed, additionally, I had to also set c->is_draining` to 1 for it to fix. I will still have to fully refactor the code to use mg_http_serve_dir but I will next weekend! 😄

closed 2023-01-09 by diamante0018